pybamm-team/PyBaMM #5: Add DAE option to solver

tinosulzer, 2018-11-06:
Allow solver.py to take `integrator="DAE"` (e.g. from SUNDIALS)
- [ ] Import SUNDIALS (or other) in Dockerfile 
- [ ] Tell the solver which equations are ODEs and which are algebraic; I think this should be done by the model class

tinosulzer, 2019-01-17:
@martinjrobins @fcooper8472 do either of you know how to install SUNDIALS in travis?

martinjrobins, 2019-01-17:
we should be able to install the serial version of SUNDIALS through apt, see https://github.com/travis-ci/apt-package-safelist/pull/1651. So that should make it easy for testing.

martinjrobins, 2019-01-17:
then use the python interface for sundials: https://pypi.org/project/python-sundials/

We'd need to add sundials as a dependency for PyBaMM as well

martinjrobins, 2019-01-28:
might want to use scipy.odes: https://scikits-odes.readthedocs.io/en/latest/installation.html

martinjrobins, 2019-01-28:
having trouble installing scipy.odes. Raised an issue in their repo to see if they can solve it

martinjrobins, 2019-01-31:
installing scikits.odes broken with pip 19, works with pip 18, seems to be pips fault.

tinosulzer, 2019-01-31:
Ok, is "works with pip 18" good enough (with hope that pip 19 will be fixed at some point) or are we worried it will put people off?

martinjrobins, 2019-01-31:
Looks like this is a fairly recent pip issue:

https://github.com/pypa/pip/issues/6163

it is marked as a bug, so presumably will be fixed, hopefully soon!

PR #109 (2019-02-05): Issue 5 dae
# Description

Fixes #5 

- install script to install scikits.odes on travis for testing
- tests are optional (skipped if scikits.odes not installed)

new classes:
- `pybamm.OdeSolver`: base class for ode solvers
- `pybamm.DaeSolver`: base class for dae solvers
- `pybamm.ScikitsOdeSolver`:  class for ode solver using scikits.odes (cvode)
- `pybamm.ScikitsDaeSolver`:  class for dae solver using scikits.odes (ida)

new model bits:
- `model.algebraic`: list of algebraic equations
- `model.initial_conditions_ydot`: initial conditions for the time derivative of variables

## Type of cha [... cut]
closed 2019-02-05 by tinosulzer