pybamm-team/PyBaMM #14: Increase Modularity of Models

Scottmar93, 2018-11-18:
I have pushed a branch: component_class to my fork of PyBaMM if you want an example. On that branch, I have briefly explore this potential class structure for models.  

The basic idea is to have two sets of classes: Models and SubModels each of which have common methods and variables. Then we derive other classes from these such as: 
`class SPM(Model):`
which then inherits the methods common to Model. It seems that this would make it easier to add a new type of model as we inherit all the standard methods. 

The Models are a two layer structure with

* Model
  * SPM 
  * SPMe, 
  * DFN, 
  * LeadAcid, 
  * User 
  * etc

The superclass 'Model' defines the methods and variables common to all models such as .initial_conditions() and .rhs()
The subclasses on level 2. then just define things specific to that model, e.g the domain and the type of submodel to use on that domain.

SubModels are arranged in a three layer structure:

* SubModel
  * Particle 
     * StandardParticle
     * DeformingParticle
     *  PhaseFieldParticle
  * Electrolyte 
     * NernstPlank, 
      * StefanMaxwell,
  * Electrode 
     * StandardElectrode 
     * BinderElectrode
  * CurrentCollector
      * StandardCurrentCollector

All SubModels should have a common structure so that they can be interacted with by Models in a consistent way. Therefore a superclass SubModel that contains common methods and variables is used (might just be conceptually useful as, for example, .rhs() will be different for each submodel).

We then have a different type of SubModel for each sub_domain: Particle, Electrolyte, etc. These define the methods and variables common to each of their domains such as setting initial conditions but allow for further specification. Finally, on level 3. a particular SubModel is choosen for a particular domain. Here specifics like the exact model equations on that domain are stated.

 **Motivation**
Increase modularity of the model section of the code

galvanic653960572, 2018-12-03:
What if some SubModel depends on result of another SubModel?
e.g. ThermalSubModel depends on power supplied by battery, and AgingSubModel depends on battery temperature calculated by ThermalSubModel
Idea for solution: solve separately in dependency order, transfer intermediate result between dependent SubModels

tinosulzer, 2019-01-16:
closing as close to duplicate of #50

closed 2019-01-16 by tinosulzer