pybamm-team/PyBaMM #16: use an expression tree to encode a model

martinjrobins, 2018-12-12:
After discussions, we've decided to switch to using an expression tree to encode each model and submodel (see #14). The idea is that this would decouple the different aspects of PyBaMM (e.g. model/set_parmeters/discretisation/solver) and allow for a pipeline approach, where the expression tree is mutated by each aspect in turn.

This issue is to provide an initial implementation of the expression tree so that it can start being used by the rest of the code.

## Possible class hierarchy:

1. `class Symbol`: base class, holds list of child nodes, has a (string) name
     - operators used to build up tree defined here (e.g. `symbol + symbol` returns `Addition(symbol,symbol))`
     - note: could we use anytree to provide basic tree functionality?: https://anytree.readthedocs.io/en/2.4.3/index.html
2. `class UnaryOperator(Symbol)`: provides a base class for all the unary operators (e.g. Negate)
3. `class BinaryOperator(Symbol)`: same but for binary ops
4. `class Addition(BinaryOperator)`: implementation of the addition op, 
5. `class Laplacian(UnaryOperator)`: implementation of the laplacian op.
6. many other ops....
7. `class Parameter(Symbol)`: a parameter (symbolic, no value). perhaps with units?
8. `class Variable(Symbol)`: a variable that will (eventually) be discretised over the mesh. At the moment it is purely symbolic
9. `class Vector(Symbol)`: a node representing a vector (the spatial discretisation class will add these in)
10. `class Matrix(Symbol)`: a node representing a matrix (the spatial discretisation class will add these in)

## iteration through tree

provide functionality for iterating through the tree, e.g.

```python
[node.name for node in PreOrderIter(f)] # pre-order traversal
[node.name for node in PostOrderIter(f)] # post-order traversal
[node.name for node in LevelOrderIter(f)] # level-order (i.e. breadth-first)
```
(note: these examples come from the anytree library)

martinjrobins, 2018-12-17:
Hi @tinosulzer. I thought it was a good idea to move the tree setup to the base class. In the process I've removed the references to child nodes as `left`, `right` and `child`, so you only have the generic `children` iterable. Let me know if this messes up anything you where doing.

martinjrobins, 2018-12-17:
Note:  `anytree` keeps track of parent/child nodes under the hood, so you can't add the same node to the tree twice. Hence we need to copy each node as its added to the tree (see commit 4919f4b)

martinjrobins, 2018-12-17:
Of course this copying is going to be a hassle when dealing with matrix nodes! Need to make sure the matrix is *not* copied....

tinosulzer, 2018-12-17:
> Hi @tinosulzer. I thought it was a good idea to move the tree setup to the base class. In the process I've removed the references to child nodes as `left`, `right` and `child`, so you only have the generic `children` iterable. Let me know if this messes up anything you where doing.

Sounds good, should only require small refactoring in #18 and #20 , we can finalise those once this issue is merged into master

martinjrobins, 2018-12-17:
commit 8395a2a tests that matrices added to the tree (multiple times) all reference the *same* matrix. That is, modifications to one modify all of them

martinjrobins, 2018-12-18:
I'm happy that this is a reasonable first draft of the expression tree. There are a few remaining issues, but I'll open separate issues for these:

1. class IndependentVariable(Symbol): a node representing an independent variable (space or time)
2. "A Scalar has a list of index ranges, and a value for each range". This is not implemented
3. `class MatrixVectorMultiplication`: not sure this is needed, as a `Multiplication` node with a `Matrix` and `Vector` children is easily identifiable and dealt with by the `Multiplication` node
4. `class Parameter`  has a `family` member, this is not documented as I'm not sure what it does
5. .... others?

tinosulzer, 2018-12-18:
> 1. class IndependentVariable(Symbol): a node representing an independent variable (space or time)

New issue?

> 2. "A Scalar has a list of index ranges, and a value for each range". This is not implemented

New issue?

> 3. `class MatrixVectorMultiplication`: not sure this is needed, as a `Multiplication` node with a `Matrix` and `Vector` children is easily identifiable and dealt with by the `Multiplication` node

Agreed that this is unnecessary. Will change `Multiplication` so that `@` is used if `self` is a `Matrix` (e.g. `Matrix * grad(var)` should use `@` even though `grad` is not a `Vector` object)

> 4. `class Parameter`  has a `family` member, this is not documented as I'm not sure what it does

Agree that this is unnecessary.

> 5. .... others?

martinjrobins, 2018-12-18:
Sounds good. I'll make those new issues for 1. and 2. once #16 is closed. I think the only thing we need to do then on this issue is to change evaluate to `evaluate(self, y, t)`