pybamm-team/PyBaMM #19: refactor BaseModel to hold an expression tree

martinjrobins, 2018-12-12:
See #16. 

The base model will internally hold an expression tree representing the LHS of an ode model. e.g.

```python
class BaseModel:
   def __init__(self):
         self._lhs = ... # lhs is a dictionary mapping variables to expressions
   @property
   def lhs(self):
          return self._lhs
```

Scottmar93, 2018-12-21:
Maybe I'm am looking to do more than is needed. Will this just end up being a class with three initially empty dictionaries (_rhs, _initial_conditions, and _boundary_conditions), with the model[variable] functionality and setters and getters for each of the dictionaries? Or do I need to consider the interactions with the other submodels and the expression tree? 

I have already done the first of the above and I am currently writing a test and the doc string for it.

martinjrobins, 2018-12-21:
> Maybe I'm am looking to do more than is needed. Will this just end up being a class with three initially empty dictionaries (_rhs, _initial_conditions, and _boundary_conditions), with the model[variable] functionality and setters and getters for each of the dictionaries? 

Yup, I think that is all that is needed

PR #46 (2018-12-21): Issue 19 refactor base model
# Description

Refactored BaseModel class to have:  
- empty dictionaries (as these will be set later) for _rhs, _initial_conditions, _boundary_conditions
- setters and getters for each dictionary using @property and @rhs.setter etc
- assigned dictionary behaviour so that rhs expressions can be accessed by model[" "] 

Note these changes break the reaction_diffusion tests etc but don't think we can get around that given the changes we are making (I have skipped these tests for now) 

Fixes #19 

## Type of change
- [x] Refactor

# Key checklist:

- [x] No style iss [... cut]
closed 2018-12-21 by tinosulzer