pybamm-team/PyBaMM #26: add a list of index ranges to `Scalar`, and a value for each range

martinjrobins, 2018-12-18:
"A Scalar has a list of index ranges, and a value for each range". This is not implemented

martinjrobins, 2018-12-19:
I've been having a go at this. Basically when you call the evaluate on this node, it should return a vector filled with values at the given ranges. So it might be more efficient to just use the `Vector` node rather than have the `Scalar` node have multiple values

tinosulzer, 2018-12-19:
> I've been having a go at this. Basically when you call the evaluate on this node, it should return a vector filled with values at the given ranges. So it might be more efficient to just use the `Vector` node rather than have the `Scalar` node have multiple values
> 
> @tinosulzer, you have been doing the discretisation, what do you think? Is this issue necessary or can we just use the `Vector` node?

I think it would be better to just use the `Vector` node, but the size ranges of the vector of the vector won't be known until discretisation. I've created a `Concatenation` class which might help:

1. Create a multi-domain parameter by concatenating single-valued parameters:
```python3
```

martinjrobins, 2018-12-20:
The concatenation looks nice! Does this mean that `Parameters` should only have a single domain, so that they can always be replaced by a single scalar?

tinosulzer, 2018-12-20:
Yes, in fact I think domain could just be a string (will always be either `"negative_electrode"`, `"separator"`, `"positive_electrode"` or `"whole_cell"`) rather than a list of strings

martinjrobins, 2018-12-20:
would it be useful to be able to easily define a single parameter that was valid over multiple domains, but had the same value over those domains?

tinosulzer, 2018-12-20:
Yes, that would be useful

martinjrobins, 2018-12-21:
ok. closing this as not needed.

closed 2018-12-21 by martinjrobins