pybamm-team/PyBaMM #37: Implement Parameter setting for a specific model

tinosulzer, 2018-12-19:
Building on issue #18 / PR #35 

For a particular, model, parameter values may be needed that are combinations of "raw" parameter values. For example, the total electrode length `L = Ln + Ls + Lp`, or any dimensionless parameter. 

It doesn't make sense to write these all in the csv file that gets imported, so there needs to be a class that calculates the "derived" parameters, maybe split by domain ("negative_electrode", "separator", "positive_electrode" or "whole_cell"). There may be overlap between the domains.
This class should also contain scales for non/re-dimensionalisation

For example,
```python3
class DiffusionParameterValues(BaseParameters):
    def __init__(self, base_parameters={}, optional_parameters={}):
       super().__init__(base_parameters, optional_parameters)
       
       self.scales = {"length": self.raw["Ln"] + self.raw["Ls"] + self.raw["Lp"],
                      "time": ...,
                      etc}
       self.whole_cell = {"L": self.raw["Ln"] + self.raw["Ls"] + self.raw["Lp"],
                          "Cd": self.scales["length"]**2/self.raw["D"]/self.scales["time"],
                          etc}
       self.negative_electrode = {"L": self.raw["Ln"], etc}
       self.separator = {"L": self.raw["Ls"], etc}
       self.positive_electrode = {"L": self.raw["Ln"], etc}
```
Maybe make `DiffusionParameterValues` emulate a dictionary, similarly to #19 , with keys being domain names.

Now if we want to include new parameters, say for a temperature model, we will need more scales and derived parameters, but these should be in a separate class as they might call upon `raw` values that would not necessarily be given to a `DiffusionParameterValues` class.

```python3
class TemperatureParameterValues(BaseParameters):
    def __init__(self, base_parameters={}, optional_parameters={}):
       super().__init__(base_parameters, optional_parameters)
       
       self.scales = {"Temp": self.raw["T_init"]}
       self.whole_cell = {"k": self.raw["D_temp"]/ bla bla}
       self.negative_electrode = {}
       self.separator = {}
       self.positive_electrode = {}
```
Then a `BatteryParameterValues` could multiply-inherit from `DiffusionParameterValues` and `TemperatureParameterValues`, combining the dictionary attributes from each of these (not sure how to do this)

@martinjrobins @Scottmar93 do you think this is a good approach? See [previous implementation](https://github.com/tinosulzer/PyBaMM/blob/structure/pybamm/parameters/parameter_values.py) (quite different) for an idea of which derived parameters need to be implemented.

martinjrobins, 2018-12-20:
I think it would make sense to encode the derived parameters and scales directly in the model. Otherwise you are always doing work to try and "sync-up" the models and the parameter classes

For the derived parameters in the model, could these just be sub-expressions in the model rhs? It makes the expressions slightly bigger, but I don't think this is a problem. e.g.

```python
Ln = pybamm.Parameter('Ln')
Ls = pybamm.Parameter('Ls')
L = Ln + Ls
rhs = L * some_expression
```

For the scales, a parameter node could have its own sub-expression indicating the units, the assumption being that the values in the parameter file are in the units specified by the model

```python
t = pybamm.Unit('m')
Ln2 = pybamm.Parameter('Ln2',t**2)
```

and then the model could have a `scales` dict that maps units to expressions:

```python
model.scales = {t: Ln + Ls + Lp}
```

so the `ParameterValues` class has to read in the parameter values, calculate the scales using `model.scales`, then then invert the units expression for each parameter to non-dimensionalise it.

tinosulzer, 2018-12-20:
I agree that encoding derived parameters and scales directly in the model (or submodels) is a good idea. 

Different submodels might need the same derived parameters, and we'll want to avoid having to define them separately in each submodel (this could easily lead to parameters that should be the same being defined differently). We could define all the parameters in the model and pass them down to submodels, but this might get messy. Any ideas for a way round this?

For dimensionless parameters, we've been giving new names to dimensionless parameters, so these can just be treated as derived parameters.
I think `model.scales` will be a useful dict for re-dimensionalisation though (plotting dimensional variables after we've solved a dimensionless model). Also, having units will definitely be useful for checks, e.g. in `Addition` and `Multiplication`, and possibly checking that all `Parameter` objects in the model are dimensionless.

martinjrobins, 2018-12-21:
> Different submodels might need the same derived parameters, and we'll want to avoid having to define them separately in each submodel (this could easily lead to parameters that should be the same [...]

Presumably, like in other fields, battery modelling has a number of "standard" parameters, and some models define their own parameters if necessary. I would suggest we do the same, have a few pre-defined parameters in `pybamm` (as global variables?) that are then used by the models. This doesn't stop individual models from defining their own parameters if necessary.

Note that Sympy does this under a separate module called `abc`, like this:

```python
from sympy.abc import x, theta
```

tinosulzer, 2018-12-21:
Brilliant, this is exactly what we need! I really like that this avoids having to evaluate parameter values if they haven't been put in the model.

So the task for this issue is to create a module from which we can import standard parameters at the start of any submodel; can split this up into different chemistries at a later stage if necessary.

PR #47 (2018-12-21): Issue 37 standard parameters
# Description

Add a few geometric parameters to a module `standard_parameters`. More parameters should be added to this as they appear in models, if they are standard battery parameters. This file could later be split into different files for different chemistries.
- Not sure whether/how to document a module that doesn't contain any classes?
- Could now rename `BaseParameterValues` to `ParameterValues` as that function won't need to be overloaded?

Fixes #37 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

-  [... cut]
closed 2018-12-21 by martinjrobins