pybamm-team/PyBaMM #45: overloaded binary operators on `Symbol` take numbers

martinjrobins, 2018-12-21:
see #36. It might be useful for the overloaded binary operators on`Symbol` (e.g. `__mul__`) be able to take `Numbers` (e.g. `int'). They would then implicitly convert this to a `Scalar` node.

Note: also need to add `__pow__` to this list.

tinosulzer, 2018-12-21:
`__pow__` , `__neg__` and `__abs__` also need to be added as methods in the `Symbol` class

tinosulzer, 2018-12-21:
Should the overloading be inside the `__mul__` function
```python3
def __mul__(self, other):
    if isinstance(other, Symbol):
         return Multiplication(self, other)
    elif isinstance(other, int) or isinstance(other, float):
          return Multiplication(self, Symbol(other))
``` 
or at the start of `BinaryOperators`
```python3
class BinaryOperator(self):
      def __init__(self, name, left, right):
             if isinstance(right, int) or isinstance(right, float):
                   right = Symbol(right)
```
?

martinjrobins, 2018-12-22:
I think the 2nd option would be better, as then it would work for all the binary operators. 

note that you can check if a python object is any type of number using `numbers.Number`, see https://stackoverflow.com/questions/4187185/how-can-i-check-if-my-python-object-is-a-number

note that if left or right is not a Symbol or a number, then the init should throw

note also that you will want to convert the left input to a Symbol if it is a number as well

PR #60 (2019-01-10): Issue 45 overloaded binary ops
# Description

- Allow `BinaryOperator` to take a number as `left` and/or `right` (gets converted to `Scalar`)
- Implement `__neg__`, `__abs__`, `__pow__`
- Implement `__r*__` for binary operators

Fixes #45 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
-  [... cut]
closed 2019-01-10 by martinjrobins