pybamm-team/PyBaMM #48: add a dictionary attribute `variables` to model

tinosulzer, 2018-12-21:
After discussions in #21 , it would be useful for the `BaseModel` class to have an attribute `variables` with that maps "variable name" to a variable. This dict needs to be processed by parameters and discretisation just like `rhs` and `initial_conditions`. Will then be useful for visualisation.

Scottmar93, 2018-12-21:
> BaseModel class to have an attribute variables with that maps "variable name" to a variable
 Does this mean maps a string "variable name" to an expression? 

If so, we can add the interface conditions into this variable dictionary e.g. 

```
variables = {"c": c, "N": grad(x), "G": homogeneous_reaction} 
```

Scottmar93, 2018-12-21:
On the back of your comment on #49, we probably cannot use this String to Expression dict for the purposes I was intending here. I was trying to figure out the best way to put the interface conditions into model. Perhaps we want a Symbol to Expression dict for this 
```
{G: homogenous_reaction}
```
With the symbol G appearing in the rhs expression trees

tinosulzer, 2018-12-21:
One option for defining models is
1. Define all variables (`c`, `Phi`, `csn`, `csp`, etc) and parameters
2. Define reactions and fluxes as
```python3
G = homogeneous_reaction() # G is now some expression, probably a multiplication
N = grad(c)
i = ...
etc
```
3. Define right-hand side, initial conditions, boundary conditions (can call submodels at this stage)
```python3
model.rhs = {c: div(N) + G, csn: ...}
model.initial_conditions = {c: Scalar(1)}
model.boundary_conditions = {N: ...}
```
4. Store all variables and expressions for later use
```python3
model.variables = {"c": c, "N": N, "G": G, etc}
```
This will mean that, at the solver stage, the discretised expression representing `G` will be calculated again each time it appears in `model.rhs`. But we should be able to optimise this at the discretisation or solver stage (e.g. by recognising recurring patterns in the expression tree). 

The alternative option is to first define variables, then define `G` and `N` as variables to pre-compute in step 2.
```python3
G = Variable("G", domain=["whole cell"])
N = Variable("N", domain = ["whole cell"])
self.pre_compute = {G: homogeneous_reactions(), N: grad(c)}
```
then 3 & 4 as before. Then the solver will have to know to pre-compute some variables before evaluating rhs.

Scottmar93, 2018-12-21:
With the first approach: 
```
G = homogeneous_reaction() 
```
the type of G is hard coded into the electrolyte submodel where as we want this G to change depending on if the model is DFN or SPM. We could just accept a function as an input when initialize the electrolyte model and the electrode models but we then have to unsure the input function remains the same across all the submodels. 

As you suggest in your second approach, I think that G shouldn't really belong to any one submodel alone and should instead be included within a separate pre_compute dict for the whole BaseModel. This pre_compute dict can then be added to when other patterns are observed in the discretisation stage

martinjrobins, 2018-12-21:
Note that it looks trivial to identify common sub-trees in a tree structure, so this is not a problem (you just recurse down your tree keeping a map of node ids to counts). If you had a common sub-expression `G` in your model, this can be identified easily at both the discretisation and solver stages and optimised out.

Scottmar93, 2018-12-21:
> If you had a common sub-expression G in your model, this can be identified easily at both the discretisation and solver stages and optimised out.

Okay this is good to know. So it doesn't matter too much which way we do it in terms of speed, it's just a matter of neatness. I think that avoiding passing interface_condition into every submodel is a bit neater. It also allows one to test the expression tree of submodels and interface_conditions separately

Scottmar93, 2018-12-21:
> you would have to pass the variable G to the submodel
Is the plan then to create all variables outside of the submodels and then pass them in? 

It is also possible to define Variables within each submodel. The submodels are then standalone expression trees with the exception of a Variable G which we is assign an expression in `basemodel`.

tinosulzer, 2018-12-21:
Yeah that seems like the right way to do it. Just need to make sure that, when the same variable is defined in different submodels, it is given the same name and domain each time. That way it will be discretised to the same StateVector (reading from the correct part of `y`) in each submodel

Scottmar93, 2018-12-21:
> Just need to make sure that, when the same variable is defined in different submodels, it is given the same name and domain each time

Yes that's good point. It probably is a bit safer to require that G and any other common variables are defined outside the submodels and passed in as an argument when the classes are initialised. In which case its likely easier just to assign G the expression right away as you suggested. 

So we have: 
- common variables defined at beginning of `basemodel` and passed in 
- submodel specific variables defined within `submodel`

closed 2019-01-14 by martinjrobins