pybamm-team/PyBaMM #49: Refactor StefanMaxwellDiffusion class

Scottmar93, 2018-12-21:
This issue must be completed prior to refactoring the reaction diffusion model #17

StefanMaxwellDiffusion should inherit from BaseModel so that it is a complete model in its own right. It will contain a dictionary with the set of equations in the electrolyte implemented in symbolic form employing the expression tree. 

Will looks something like:

`class StefanMaxwellDiffusion(BaseModel): 
        def __init__(self): 
            self._rhs = {"c": "epression"}
            self._initial_conditions = .... etc 
`

tinosulzer, 2018-12-21:
Yep looks good - the keys and variables of _rhs, etc should be Symbols though:
```
class StefanMaxwellDiffusion(BaseModel): 
        def __init__(self):
            c = Variable("c", domain=["whole cell"]
            N = ...
            expression = ...
            self._rhs = {c: expression}
            self._initial_conditions = {c: ...}
            self._boundary_conditions = {N: (left bc, right bc)}
            self._variables = {"c": c, "N": N}
```

Scottmar93, 2018-12-21:
I am unsure what tests are appropriate to run on this class. 

I have push what I have done so far if anyone wants to take a look/take over

tinosulzer, 2018-12-21:
Cheers Scott, looking good. Very basic tests could be just making sure that `model = pybamm.ReactionDiffusionModel()` doesn't throw any errors (e.g. with `G = Scalar(1)`), then seeing if you can process the model with appropriate parameters and discretisation, and finally something along the lines of the functions currently in `test_reaction_diffusion.py` to make sure that the input/output has the right shape and the model behaves physically as expected

PR #61 (2019-01-14): Issue 49 refactor stefan maxwell diffusion
# Description

Refactored StefanMaxwellDiffusion class to create an expression tree. Only a simple test to see if the tree is created without errors is implemented. This will need to be revisited when: 
- power operator is implemented for class Parameter
- can write `rhs = {c: - grad(c)}` instead of `rhs = {c: Symbol(0) - grad(c)}` 
- dimensionless parameters are created and set by the parameter class  (I was unsure how to set a parameter that is different in different regions). 

Fixes #49 

## Type of change

Please delete options that are not relev [... cut]
closed 2019-01-14 by tinosulzer