pybamm-team/PyBaMM #50: Combining submodels

Scottmar93, 2018-12-21:
Following from #49 

If each submodel (e.g. StefanMaxwellDiffusion) is a Model in its own right, we should have a function which allows for the simple concatenation of two submodels e.g. the functionality: 
```
basemodel.create_from_submodels([submodelA, submodelB, submodelC, etc])
```
This would concatenate the rhs, initial_conditions, boundary_conditions, variables dictionaries. 

It is possible that overlapping variable names may become an issue (e.g. concentration in negative and positive particle if within the particle expression they are simply called c). By associating a unique name to each submodel when it is initially defined, this could be avoided.

Scottmar93, 2018-12-21:
Adding to this. It may be good to have a parent class Model which contains the basic setting, getting, and empty required dictionaries. We could then have two subclasses: BaseModel and SubModel. The BaseModel would have the additional method of combining SubModels. The SubModel class would contain an additional 'name' attribute.

martinjrobins, 2018-12-21:
You could probably implement the `__init__` of the `BaseModel` to take an argument that was a list of models, then just create the concatenated model there, that way you ensure that `basemodel` is starting out empty

Scottmar93, 2018-12-21:
> How about an update function that can update a model's rhs, initial_conditions, boundary_conditions and variables using a submodel?

I think this is a good idea. I think the `create_from_submodels` function I defined above would basically just loop through the submodels calling that update function. I think we should have both an update function and a create from submodels function. One offers the user the option of easily appending some additional physics to a model whilst the other offers a clean way to piece together many submodels at once without needing to write the update function every time.

tinosulzer, 2018-12-21:
Yep sounds good.

tinosulzer, 2019-02-04:
Closed by #108

closed 2019-02-04 by tinosulzer