pybamm-team/PyBaMM #51: Refactor HomogenousReactions class

Scottmar93, 2018-12-21:
This is a pre-requisite for #17 

Refactor this class to that it simply returns an symbolic expression for the Homogeneous reactions. 

Should this really be a class? A function is probably sufficient

tinosulzer, 2018-12-21:
Agree that a function should be sufficient. 

Another prerequisite for this is defining an appropriate current function `icell(t)` (note that we already have a class for the independent variable `t` in the expression tree).

Scottmar93, 2019-01-14:
requires #67 so that can have: 
`G = (Scalar(1,"negative electrode"), Scalar(0, "separator"), Scalar(-1,"positive electrode") `
Where `G` is the flux between electrode and electrolyte.

Scottmar93, 2019-01-14:
requires #68 for time dependent homogeneous reaction rate.

PR #107 (2019-01-31): Issue 51 homogeneous reactions
# Description

Fixes #51 
Just a simple case where the current is constant with a value of 1. Will need to change to a time-dependent current once we have a current function

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] Tests added that prove fix is e [... cut]
closed 2019-01-31 by martinjrobins