pybamm-team/PyBaMM #63: discritisation of boundary conditions

martinjrobins, 2019-01-10:
I was just looking at the discretisation of the boundary conditions. At the moment this seems to be mainly done in the `gradient` and `divergence` functions of `MatrixVectorDiscretisation`, which both do this:

```python
       # Add boundary conditions if defined
        if symbol.id in boundary_conditions:
            lbc, rbc = boundary_conditions[symbol.id]
            discretised_symbol = self.concatenate(lbc, discretised_symbol, rbc)
        gradient_matrix = self.gradient_matrix(domain)
        return gradient_matrix * discretised_symbol
```
So it looks like extra ghost nodes are added to the current state vector to implement the bc. Should this be done instead when the `pybamm.Variable` node is processed in `process_symbol`? Also, the way the boundary conditions are implemented would depend on the method, so this should probably be done in `FiniteVolumeDiscretisation`? @tinosulzer?

tinosulzer, 2019-01-10:
Yes, this is quite hacky as currently implemented: for example, there will be errors with matrix sizes if no boundary conditions are given. Moving to `FiniteVolumeDiscretisation` seems sensible for now but we should discuss how to do this properly

martinjrobins, 2019-01-14:
After meeting (14/1/19):
- [ ] add checks in Discretisation that there are either Dirichlet or Neumann bcs for each fundamental vector
- [ ] add support for Dirichlet bc.  concatenate appropriate ghost nodes on StateVector

PR #92 (2019-01-17): Issue 63 discretise bcs
# Description

Allow processing of Dirichlet boundary conditions in Finite Volumes. Need ghost nodes for this so:
- add function `add_ghost_node` to `FiniteVolumeDiscretisation`
- update boundary condition setter in `FiniteVolumeDiscretisation.gradient`
- add function `combine_submeshes` to `FiniteVolumeMacroMesh` and use this when calculating grad and div matrices (this would also allow calculating gradient/divergence for variables with several subdomains)
- move functions from `MatrixVectorDiscretisation` into `FiniteVolumeDiscretisation` as they are discretisation-de [... cut]
closed 2019-01-17 by martinjrobins