pybamm-team/PyBaMM #68: add a `Function` node to expression tree

martinjrobins, 2019-01-14:
for `Parameters` that vary with time, user puts in the name of the function (functions defined in standard_parameters). Then the parameter class replaces `Parameter` with `Function` of that name

martinjrobins, 2019-01-28:
the python function will take two arguments:
1. the evaluation of its child (ie. a state vector)
2. the current time

Note: need to update:
https://github.com/tinosulzer/PyBaMM/blob/b75cce796a59e25f723e2cf7eed11068d2bb3fc4/pybamm/expression_tree/symbol.py#L288

fcooper8472, 2019-01-29:
`Function` is now implemented as unary operator which addresses part of this issue.

The other part, as I understand it, is allowing a function pre-defined in a `.py` file to be specified in a `csv` parameter file.  I envisage something along these lines:

* Put a `.py` file in the same directory as the `csv` parameter file
* Specify the name of the function in the `csv` file
* Enforce that the name of the python file matches the name of the function
* Dynamically load the function when symbols are being processed

In this way the `csv` would look something like
```
Name,Value,Units,Meaning,Reference,Notes
# Empty rows and rows starting with ‘#’ will be ignored,,,,,
,,,,,
# Functions,,,,,
Test function,test_function,,,,
,,,,,
```
where `test_function` is a function living inside `test_function.py` in the same directory as the `csv` file.  Does this sound reasonable as a starting point?

martinjrobins, 2019-01-29:
Sounds good. So `Test function` in this case will be the name of the Parameter node, which would be replaced by a `Function` node containing the `test_function` function within the `test_function.py` file, is that right?

fcooper8472, 2019-01-29:
That's my understanding of the desired behaviour, yes. @tinosulzer feel free to add anything to this!

tinosulzer, 2019-01-30:
Yes, that sounds right. 
See below for examples of functions that would need to be implemented for the lead-acid models. 
[image]
For functions like D(c), which just consists of numbers and c, then the approach above would work perfectly. 
However, for functions like rho(c) that also depend on the parameters M_w, M_e, V_e and V_w, the approach would have to be slightly different, because the parameters M_w, M_e, V_e and V_w need to also be assigned values at the "set parameters" stage. I'm not sure what the best way of doing this is. 
One lazy workaround is to only implement functions like rho(c) in [`standard_parameters.py`](https://github.com/tinosulzer/PyBaMM/blob/87785355fd11225a031c1c4b4a4f663e4de95403/pybamm/parameters/standard_parameters.py), where we can define them as expression trees, but that makes it harder for the user to input their own functions

tinosulzer, 2019-02-01:
Have thought about this a bit more, I think we can differentiate between "experimental functions" (measured from experimental data - D, chi, kappa, mu, U) that can be provided externally as you've suggested above and "model functions" (derived from physics - rho, cw, j0) that are really part of the model and should be defined in the model/submodel classes. So I think the exact functionality you've suggested above would be what we need for this issue

fcooper8472, 2019-02-01:
Ok, great.  I'll get to work on that.  There doesn't seem to be a 'clean' way of dynamically loading a python function, but it's certainly doable.

tinosulzer, 2019-02-04:
A small edit on the above after discussion today (see suggested unit test): the model will want to create a `Function` object in the form
```python3
c = Variable("c")
D = Function(c, "diffusivity")
```
On setting parameters, D should become a `Function(c, diffusivity)` where `diffusivity` is a python function defined in `input/parameters/lead-acid/diffusivity.py`

fcooper8472, 2019-02-14:
Progress on this:
 * New class `FunctionParameter` similar to `Parameter` but with a child
 * Functionality in `util.py` to `load_function`.  This takes a string that is either an absolute path to a `.py` file containing the function to load, or a relative path in which case the file is searched for everywhere in the PyBaMM directory.
 * Example file in `test_parameters/data/process_symbol_test_function.py`, with the following test showing current usage:
```python
    def test_process_function_parameter(self):
        parameter_values = pybamm.ParameterValues(
            {"a": 3, "func": 'process_symbol_test_function.py'})

        # process parameter
        a = pybamm.Parameter("a")
        processed_a = parameter_values.process_symbol(a)
        self.assertIsInstance(processed_a, pybamm.Scalar)
        self.assertEqual(processed_a.value, 3)

        # process function
        func = pybamm.FunctionParameter("func", a)
        processed_func = parameter_values.process_symbol(func)
        self.assertIsInstance(processed_func, pybamm.Function)
        self.assertEqual(processed_func.evaluate(), 369)
```

Does this look OK @tinosulzer @martinjrobins @Scottmar93 ?

tinosulzer, 2019-02-14:
Looks great! I can see what you meant about dynamically loading a function being messy. Only comment here would be that `FunctionParameter` should take arguments the other way round to be more consistent with `Function`

martinjrobins, 2019-02-14:
looks good to me. make a PR and we can have a look at the code too.

fcooper8472, 2019-02-14:
@tinosulzer `FunctionParameter` takes `self, name, child` which is the same as `UnaryOperator` that it inherits from.

I suggest keeping it like this for now and open a new issue for swapping, which might be a wider change than just `FunctionParameter`

PR #147 (2019-02-15): Issue 68 function node
# Description

#68 Implements `Function` and `FuntionParamter` as well as machinery for dynamically loading a function given string representing the filename.

Fixes #68

## Type of change

Please delete options that are not relevant.

- [ ] New feature (non-breaking change which adds functionality)

# Key checklist:

- [ ] No style issues: `$ flake8`
- [ ] All tests pass: `$ python run-tests.py --unit`
- [ ] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tests.py --quick`.

## Further checks:  [... cut]
closed 2019-02-15 by martinjrobins