pybamm-team/PyBaMM #69: testing TODO

martinjrobins, 2019-01-14:
- general convergence test
- test convergence of complicated model to simple
- analytical solutions to specific models
- method of manufactured solutions

tinosulzer, 2019-04-14:
Have started adding manufactured solution, to do:
- [ ] explain MMS in `ManufacturedSolution` docstring
- [ ] finish testing the behaviour of `ManufacturedSolution` on known (easy) models
- [ ] set up standard MMS convergence test for models (ideally after #279 merged)
- [ ] figure out how to do explicit integral: code up some very simple cases, or always provide explicitly?

martinjrobins, 2019-09-30:
@tinosulzer : is this finished now?

tinosulzer, 2019-09-30:
I was working on using Method of Manufactured Solutions to check convergence, there were some problems with it and I don't think this is a priority any more and if we wanted to we could just check error relative to the finest mesh. There might be some stuff to merge in this branch though so I will check

tinosulzer, 2020-05-06:
Unlikely to be done any time soon; closing

closed 2020-05-06 by tinosulzer