pybamm-team/PyBaMM #76: Bug when using model.rhs.evaluate()

Scottmar93, 2019-01-15:
There is an issue using the `model.rhs.evaluate()` function which depends on the form of the flux `N_e` in the electrolyte model which is indicated by an arrow in the code below: 

```
       epsilon = pybamm.standard_parameters.epsilon_s  
        b = pybamm.standard_parameters.b
        delta = pybamm.standard_parameters.delta
        nu = pybamm.standard_parameters.nu
        t_plus = pybamm.standard_parameters.t_plus
        ce0 = pybamm.standard_parameters.ce0

        electrolyte_domain = ["whole cell"]
        c_e = pybamm.Variable("c_e", electrolyte_domain)

        N_e = -pybamm.grad(c_e)  <--------------------------------------------------(1)
        self.rhs = {
            c_e: -pybamm.div(N_e) / delta / epsilon + nu * (1 - t_plus) * G
        }  # / delta / epsilon + nu * (1 - t_plus) * G
        self.initial_conditions = {c_e: ce0}
        self.boundary_conditions = {
            N_e: {"left": pybamm.Scalar(0), "right": pybamm.Scalar(0)}
        } 
```
Within `test_electrolyte.py` I have
```
        G = pybamm.Scalar(1)
        model = pybamm.electrolyte.StefanMaxwellDiffusion(G)

        param = pybamm.ParameterValues(
            "input/parameters/lithium-ion/parameters/LCO.csv"
        )

        param.process_model(model)

        mesh = pybamm.FiniteVolumeMacroMesh(param, 2)
        disc = pybamm.FiniteVolumeDiscretisation(mesh)
        disc.process_model(model)

        y0 = model.initial_conditions
        model.rhs.evaluate(None, y0) <---------------------------------------(2)
```
However, if i change (1) to `N_e = epsilon * pybamm.grad(c_e)` , there is an error at (2). Link to the failed test output: https://gist.github.com/Scottmar93/b7bcbff685383d9a5dceaca630f3aa40

Scottmar93, 2019-01-15:
Note that the parameter `epsilon` is used elsewhere so its not a parameter import error

tinosulzer, 2019-01-15:
Have found the issue here. 
When discretising spatial operators, we look in `model.boundary_conditions` to see if the variable/expression whose gradient/divergence is being taken is in the keys. 
Here, the key in `model.boundary_conditions` is `Parameter("epsilon_s")*Gradient(c)` (the original form of `N_e`), but the variable `N_e` has been through `parameters.process_symbol` and so it is now `Scalar(1)*Gradient(c)`. 
This means that `N_e` is no longer being recognised as one of the keys of `model.boundary_conditions`, so the discretised vector is too short by 2.
 
Easiest solution seems to be to process parameters for `boundary_conditions.keys()` so that `Parameter("epsilon_s")` get converted to `Scalar(1)` there as well.

There also need to be more informative error messages when there is a shape mismatch in the discretisation (i.e. error should appear in discretisation, not when trying to evaluate in solver), so that errors like these are easier to identify. I will include this as part of #63

PR #89 (2019-01-17): Issue 76 boundary conditions bug
# Description

Process parameters for boundary conditions so that they are recognised when discretising if a Parameter changes to a Scalar (see #76 for clearer description).

Fixes #76 

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is commented, particularly in hard-to-un [... cut]
closed 2019-01-17 by tinosulzer