pybamm-team/PyBaMM #80: allow concatenation of variables and scalars

martinjrobins, 2019-01-15:
From @tinosulzer  in #77:

It would useful to allow concatenation of variables and scalars, to deal with a case like the following:
Interfacial current G is some function in the positive and negative electrodes, and zero in the separator. In the model, this would probably be coded as

Gn = cn ** 2
Gs = Scalar(0)
Gp = cp ** 3
G = Concatenation(Gn, Gs, Gp)
Then G should get discretised to a concatenation of an expression tree with a StateVector, representing cn, for the negative electrode, a Vector of zeros with the right shape, and an expression tree StateVector, representing cp, for the positive electrode.

martinjrobins, 2019-01-15:
To implement this, I would propose that when the discretisation class come across a `Concatenate`, then it:
   1. checks if any of the children expressions vary with time (i.e. have variables in them)
   2. if any children *don't* vary with time, then work out the domain of that child and the value it evaluates to, and replace the child with a Vector of the right shape
   3. if all children don't vary with time, then replace the concatenate by a concatenated Vector
   4. if any children vary with time, replace the Concatentate with a NumpyConcatenate

tinosulzer, 2019-01-15:
For 1., it might be useful to differentiate between functions that need a `t` and `y` input in `evaluate`, and those that just have a `value` (Scalars, Vectors, Matrices). 
e.g. a `Multiplication` of two nodes that both have a value could itself get a value.
This is stepping a bit into the territory of issue #54 though

martinjrobins, 2019-01-15:
agreed. Whichever is done first, this one or #54, should implement a test to see if you can evaluate an expression without `t` or `y`, then we could use this for both.

tinosulzer, 2019-01-17:
this needs to be done before #51 (which in turn needs to be done before #17)

PR #94 (2019-01-19): Issue 80 concat var scalar
# Description

To allow concatenation of more than just Scalars, made a new concatenation class `DomainConcatenation` that concatenates vectors (evaluated by child symbols) according to their domains, respecting both the ordering in KNOWN_DOMAINS, and the number of points within each domain. Just like the base `Concatenation`, it is assumed that the domains of the children are all distinct

As part of this work, added a `is_constant` function to `Symbol` that tests if evaluating that symbol depends on either y or t (this will also help with #54 and #55)

Fixes #80 

 [... cut]
closed 2019-01-19 by martinjrobins