pybamm-team/PyBaMM #82: Give domain to all expressions

tinosulzer, 2019-01-15:
Currently, only `Variable`, `Parameter` and `Scalar` nodes have domains. It would be useful to be able to evaluate the domain of an arbitrary node, e.g. `2*var` which is a Multiplication.
Proposed edit is:
- For a `UnaryOperator`, inherit domain from child (can be `None`)
- For a `BinaryOperator`, check that children have the same domain (or either or both `None`) and inherit that domain
- Additionally, check that key and value have the same domain in `model.rhs` and `model.initial_conditions` (via the respective setters in `BaseModel`)

This will help with #78 , and allow us to not have to pass `domain` as an argument in the discretisation functions.
@martinjrobins thoughts on this?

martinjrobins, 2019-01-15:
Yes, this would be really good, and would help with #80 as well.

presumably if you `Concatenate` two or more expressions you add their domains into a combined list? How do you know when you get to "whole cell" though?

tinosulzer, 2019-01-15:
Yes, agree with that approach for Concatenate. 
For now we could treat ["negative electrode", "separator", "positive electrode"] and ["whole cell"] as equal, but maybe ["whole cell"] should be deprecated

tinosulzer, 2019-01-15:
What is the intended functionality of having a separate `Domain` class? Could we make `domain` (+ setters and getters) an attribute of `Symbol`? I'm struggling with multiple inheritance ...

martinjrobins, 2019-01-15:
if everything is going to have a domain, then it makes sense to merge the `Domain` class with the base `Symbol` class, so I'm happy if you want to do this rather than have multiple inheritance

PR #83 (2019-01-16): Issue 82 domains
# Description

Add domain to all nodes, with inheritance from children nodes for operators and checks that domain combinations make sense
- Merge `Domain` class into `Symbol` class (default domain is `[]`)
- Unary operator inherits from child
- Binary operator checks that children have same (or empty) domain and inherits 
- Concatenation checks that children have disjoint domains and inherits
- BaseModel checks that variable/equation pairs have the same domain in `rhs`. also in `initial_conditions`.
- Separate discretised rhs/initial_conditions from concatenated rhs/initial_co [... cut]
closed 2019-01-16 by tinosulzer