pybamm-team/PyBaMM #93: Create Geometry class and refactor meshing

Scottmar93, 2019-01-17:
**Summary**
Create a class `Geometry` of the form 
```
    class Geometry(model_type="DFN"): 
              self.macro = {"negative electrode": {"x": {"min": 0, "max": ln}}, 
                              "separator":{"x": {"min":ln,"max":ln+ls}}, ......} 
              self.micro = {"negative particle": {"r":{"min":0, "max":1}}, ....} 
              self.model=model_type

     def process_geometry():

     def mesh_geometry():

```
`Geometry`  should:   
  - hold a dict of macroscale domains for the spatial variables and their min max values (this can be extended to 3D by adding a "y" and a "z") so all the problem dimension information is all held here. 
- hold a dict of microscale domains for the spatial variables and their min max values 
- have a process geometry function which acts in the same way as process model to convert parameters to numbers 
- have a mesh function which modifies the geometry class and replaces the macro and micro dictionaries by their discritised versions i.e.
```
 macro = {"negative electrode": [array],  # or a matrix if geometry was 2D or 3D (or a Fenics Mesh)
                              "separator": [array]
                "whole cell": [array]
```
(this mesh function would need to take in a dict of the types of meshes desired on each domain and the number of grid points desired. If one number is stated for the macro points it will divide them properly among "negative electrode" "separator" etc.)
- after meshing, `Geometry` will also store the grid points on each of the domains in a structure similar to the macro and micro entries. 
- a flag `model_type` can be also stored to disinguish between SPMe and DFN which have the same macro and micro geometry before meshing but the micro mesh is an array for the SPMe (is 1D) and a matrix for the DFN(also multplied by the macroscale mesh). (This could just be a case of having a kwarg model="SPMe" when creating a Geometry object. Or we could feed the model in when meshing the geometry). 

Where we previously had a Mesh class, we now have mesh functions only called upon by `Geometry`. These will be things like `uniform_mesh_1D` which takes in a set of domains and the number of points on that domain and outputs an array of values. 

 **Motivation**
Separate the statement of the cell geometry from the meshing. Currently the geometry is stated within the `FiniteVolumeMeshClass` and would therefore need to be restated each time a new `Mesh` class is made. It also means that when we add particles to the setting that we need a new mesh class despite there a 1D uniform mesh already being written within the `FiniteVolumeMesh` class. 

The proposed method puts all the geometric features that are common to all meshes in a class and reduces the work each mesh function must do to just taking in the limits of a domain (or set of domains) and the number of points needed on that domain and outputting an array. I think we would also be able to pass Fenics meshes around instead of arrays within this structure.

tinosulzer, 2019-01-17:
This all seems very sensible, cheers Scott. A few minor points:
- `Geometry` could extend the `dict` class so that we can just define things like `self["negative electrode"]` instead of `self.macro["negative electrode"]` (like `ParameterValues` and `BaseMesh` do already)
- `process_geometry` should be a method in `ParameterValues`
- `mesh_geometry` should be a method in a discretisation class. For example, `FiniteVolumeDiscretisation` could have a method `mesh_geometry` that does essentially what `FiniteVolumeSubmesh` does. This would actually help a lot as it would avoid the potential issue of users using the wrong mesh/discretisation pair.

So `Geometry` just states the geometry and then passes it on to setting parameter values and discretising.

I don't think `Geometry` should have a `model` attribute - we should be able to make geometry model-independent and make the discretisation figure out what to do with a particular model/geometry pair.

Scottmar93, 2019-01-17:
note from discussions: 
- keep the meshes as classes that have nodes, number of points, edges etc instead of arrays

tinosulzer, 2019-01-25:
Note changes from #98 will affect this slightly

closed 2019-02-11 by Scottmar93