pybamm-team/PyBaMM #101: Backwards compatibility with other versions of Python

tinosulzer, 2019-01-27:


fcooper8472, 2019-02-14:
This has caused a number of issues.

* Python 3.4: `Could not find a version that satisfies the requirement pandas>=0.23 (from pybamm==0.0.0)`
* Pythons 3.5 and 3.7 fall over on the scikits solvers

I'll look into them, but any immediate ideas @martinjrobins ?

tinosulzer, 2019-02-14:
AFAIK the pandas version number is arbitrary, everything would probably still work with earlier versions

fcooper8472, 2019-02-14:
Ok, also looks as though the `@` operator was not introduced until Python 3.5.  Anyone know if you can pull it in in 3.4?

tinosulzer, 2019-02-15:
looks like `np.dot` has almost the same functionality as long as we don't try to multiply 3D matrices

fcooper8472, 2019-02-15:
If we're at the point of ditching more modern language features, though: are we serving a big constituency by supporting python 3.4?

I notice that 3.4 is the default python3 on Ubuntu 14.04 LTS, but that reaches end of life in April this year.  Ubuntu 16.04 LTS gives you 3.5 by default.  If you're installing with anaconda in any case you're getting the latest, which likely covers anyone using windows.

I would be as happy to not support 3.4 as I would be to replace `@` with `np.dot`, but happy to be overruled.

martinjrobins, 2019-02-15:
I'm happy to require python 3.5 and up

fcooper8472, 2019-02-16:
@martinjrobins any ideas on the ODES related failures? I'm unable to reproduce them outside of travis... everything seems to be installed and set up correctly.  Can't think of any reason that it would work on 3.6 but not 3.5 or 3.7...

martinjrobins, 2019-02-17:
The README.md for odes.scikit says its for Python 3.3-3.5, so they might not have tested on 3.7. But it doesn't explain why 3.5 is not working. Maybe you can have a look at their travis setup, make sure we arn't missing anything?

tinosulzer, 2019-02-22:
Will need to update `README.md` and `CONTRIBUTING.md` to say we only support Python3.5+

fcooper8472, 2019-02-25:
Ok, I've tried using the sundials install scripts used by [ODEs](https://github.com/bmcage/odes), and we're getting the same errors.

I'll open an issue over at ODEs and see if there's something obvious we're doing wrong.

fcooper8472, 2019-02-25:
See https://github.com/bmcage/odes/issues/92

fcooper8472, 2019-03-07:
https://travis-ci.org/tinosulzer/PyBaMM/jobs/503052350

closed 2019-05-11 by tinosulzer