pybamm-team/PyBaMM #102: create a discretisation class for firedrake

martinjrobins, 2019-01-28:
This will supply a FEM discretisation option for PyBamm

Firedrake is located here https://www.firedrakeproject.org/index.html

tinosulzer, 2019-02-05:
Spoke to someone who is familiar with Firedrake about installation. They have a long [installation script](https://www.firedrakeproject.org/download.html) that installs PETsc and a bunch of other libraries in a virtual environment, and this virtualenv must be activated to use Firedrake. 

He suggests our options are 
1. Install Firedrake as recommended, activate the virtualenv and then `pip install pybamm` (and maybe sundials and hope that raises no conflicts?)
2. Edit our own copy of the firedrake install script so that it automatically installs pybamm (and maybe sundials) in the virtualenv as well

Firedrake installation is looooong so we should have an option to disable on Travis unless explicitly required - maybe default is to only install firedrake and run firedrake tests overnight, then is it possible to trigger different options from a commit/pull request, similarly to [ci skip]?

martinjrobins, 2019-02-05:
Both options seem bad. This is why I don't like libraries that provide their own custom
install scripts! 

It seems likely that someone would want to install pybamm (into a virtualenv), play with
it a bit. And *then* want to install firedrake. What you really want is a install script
for firedrake that installs into the *current* virtualenv, not create a new one.

I'll have a look at their script. Probably option 1 is the way to go, because no doubt
the firedrake ppl will change the install script at some point in the future, and we
don't want to have to propagate those changes.

Another option is to use something else. I found this:

https://github.com/kinnala/scikit-fem

This sacrifices speed for easy-to-use. What are your thoughts on this? If performance
(and distributed computing) isn't a priority for us then something easy-to-use might be
preferable. This uses numpy as well, so might be more compatible with what we are doing
currently.

Quoting Valentin Sulzer (2019-02-05 15:07:01)

martinjrobins, 2019-02-05:
great, I'll swap to using `scikit-fem` for this issue

tinosulzer, 2019-02-07:
@Scottmar93 is making  changes to `base_discretisation`, might be good to wait until that is done before starting this (if not started already)

martinjrobins, 2019-02-07:
OK, I've started but mainly doing the mesh classes so far

closed 2019-05-07 by martinjrobins