pybamm-team/PyBaMM #110: smarter domain concatenation

martinjrobins, 2019-02-04:
At the moment, the DomainConcatentation class will convert any Scalar children to vectors. This should be generalised, so that if the `evaluate` of its children return a scalar, this is appropriately mapped to a vector at run-time (i.e. for time dependent functions)

tinosulzer, 2019-02-20:
This issue should:
- unskip the test from #149 
- remove the warning from #124 
- implement #155 
- test that concatenation of 2D matrices works as expected, not just 1D vectors (e.g. the matrix representing c as a function of space and time)

martinjrobins, 2019-02-28:
discussed this in meeting, see result here: https://github.com/tinosulzer/PyBaMM/pull/163#issuecomment-468256714

PR #163 (2019-03-01): Issue 110 domain concatenation
# Description

**To be reviewed after #156 **
Simplifies the `DomainConcatenation` and allows it to work with a range of classes. 
- [x] Allow concatenation of StateVector objects 
- [x] Allow concatenation of 2D objects (e.g. solutions of the model in different domains) - a little bit hacky as currently written, with lots of if statements

Fixes #110 and #155 

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All  [... cut]
closed 2019-03-01 by martinjrobins