pybamm-team/PyBaMM #111: integration expression tree node

martinjrobins, 2019-02-04:
need to be able to define integrations over space of the form:
$$
f(x) = \int_0^x  f(x')dx'
$$

These will be replaced by a matrix-vector multiply by the discretisation

tinosulzer, 2019-02-16:
@Scottmar93 should the integral for a spherical domain have a factor of 2*pi or not?

Scottmar93, 2019-02-20:
> should the integral for a spherical domain have a factor of 2*pi or not?

I would put it in. 

If we are just integrating to determine conservation of mass it doesn't really matter. Total lithium-ions in the electrolyte should be conserved. Therefore total lithium in the electrode particles is conserved (and 2pi factor is in both negative and positive electrode particle integrals). 

If we want to output the total lithium in each electrode independently then it is probably clearer to put in the 2pi so we can say its the amount in the particles instead of the amount in a portion of the particle (I don't know how to refer to this properly).

PR #186 (2019-03-07): Issue 111 integral node
# Description

Node to calculate the definite integral, needed for some of the reduced-order models
Fixes #111 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] Tests added that prove fix is effective or that feature works
- [ ] Any dependent changes have been  [... cut]
closed 2019-03-07 by tinosulzer