pybamm-team/PyBaMM #129: enable per-equation discretiation

martinjrobins, 2019-02-11:
depends on #128 

dict of {variable: discretisation} passed as input to base discretisation class. 

The discretiation class passed in the dict handles things like gradient/divergence discretisation + broadcasting

Scottmar93, 2019-02-13:
Should `compute_diffusivity` which finds the value of the diffusion coefficient at the locations between nodes be specific to the 'FiniteVolume' method? If we were to implement a `FiniteDifference` method, we could still call the same `compute_diffusivity` function to find the the values at the locations between nodes (even though the nodes are now shifted). Not sure of the requirements for `FiniteElement`

Scottmar93, 2019-02-14:
@tinosulzer suggested that instead of providing spatial methods per equation, we give a spatial_method to each variable.  

With the per equation method we encounter a problem when we try discretise the dict of variables which is of the form: 
```
model.variables = {"c", c, "N", N} 
```
There is no eqn key id to determine which spatial method we should use.

tinosulzer, 2019-02-14:
I would also suggest giving `spatial_method` the variable names as keys, e.g. if we define 
```python3
c = Variable("concentration", domain=whole_cell)
```
then the spatial method could have keys 
```python3
spatial_method = {"concentration": ..., ...}
```
We could even make more specific variable classes like `Concentration` , `Potential`, etc that are subclasses of `Variable` but give the right name

martinjrobins, 2019-02-14:
that sounds reasonable. I guess when you pick a certain set of basis functions, you are picking them to represent a certain variable, rather than an equation, so I think it makes sense.

martinjrobins, 2019-02-14:
> then the spatial method could have keys
> 
> ```python
> spatial_method = {"concentration": ..., ...}
> ```
> 
This seems inconsistent with what we are doing for model.rhs and model.algebraic? We have used the variable object as the key for these, why not spatial_method as well?

tinosulzer, 2019-02-14:
Yes, good point, better to keep it consistent.

PR #152 (2019-02-18): Issue 129 enable per equation discretisation
# Description
A per **domain** implementation of the spatial methods. This is instead of per equation because of some issues described below. 

This can be made sufficient for different finite elements by adding a polynomial type property to the variables we are solving for. It won't allow for finite volumes and finite elements both on the macroscale but we don't foresee that being something which is needed. 

Fixes #129 

# Details
I have implemented a per domain spatial method assignment. The main reason to do this instead of per equation is that  [... cut]
closed 2019-02-18 by Scottmar93