pybamm-team/PyBaMM #130: Windows and MacOS support

fcooper8472, 2019-02-11:
Check whether we can get everything working on Windows and MacOS, and add these to travis

tinosulzer, 2019-02-11:
Would be good to do #101 at the same time

ferranbrosa, 2019-05-29:
I am working on Windows, so I can help with that. The installation is pretty much as described except for a minor change when activating the virtual environment.

Also, in .gitignore we should add env/ as that is the folder created in Windows (I think).

ferranbrosa, 2019-05-29:
After installing it in Windows I found that the tests failed (and I haven't changed anything from the most recent version). It seems that all errors are because it fails to import the test module:

ImportError: Failed to import test module: test_util

Also, and probably related to this, I tried running compare_lithium_ion.py and I get an error saying:

FileNotFoundError: [Errno 2] File b'C:\\Users\\Planel_F\\OneDrive - University of Warwick\\PyBaMM\\examples\\scripts\\input\\parameters\\lithium-ion\\mcmb2528_lif6-in-ecdmc_lico2_parameters_Dualfoil.csv' does not exist: b'C:\\Users\\Planel_F\\OneDrive - University of Warwick\\PyBaMM\\examples\\scripts\\input\\parameters\\lithium-ion\\mcmb2528_lif6-in-ecdmc_lico2_parameters_Dualfoil.csv'

I tried copying the input folder on scripts, but got a similar error with another folder. I think the problem is that it sticks the path to the current file in front of the relative path, instead of sticking the path to PyBaMM root folder. I guess that could be an OS issue and also the cause for all tests failing. 

Does this have a quick fix? If so, I can try it in my local copy and see if it works.

tinosulzer, 2019-05-29:
Hey @ferranbrosa , glad you're getting involved, would be great to sort out installation on Windows. I don't think anything has been done on this yet.

For the tests, I'm not sure why it's not working, but my best guess is that Windows doesn't like the forward slash on [line 30 of `run_tests.py`](https://github.com/pybamm-team/PyBaMM/blob/18632d2d2df68e3a5c14dae746c9a4e2de710142/run-tests.py#L30). You'll need to change this to using `os.path.join`. There are a few other places where the forward slash is hardcoded, e.g. in [`base_models.py`](https://github.com/pybamm-team/PyBaMM/blob/18632d2d2df68e3a5c14dae746c9a4e2de710142/pybamm/models/base_models.py#L628), possibly in a few other places too.

For the `compare_lithium_ion.py` script, try running from the root folder (`python3 examples/scripts/compare_lithium_ion.py`) - at least it should find it there, but it will probably fall over because of the forward slash issue

martinjrobins, 2019-05-29:
Thanks @ferranbrosa, as Tino says, feel free to make a PR to add any changes to the installation for windows, and the bugs that you find. I'll add you to this issue

ferranbrosa, 2019-05-30:
The tests still fail even though I have changed the path definition using `os.path.join`, however I tried running `SPMe.py` from the root folder and it worked fine. It could be that the test error is caused by something else...

tinosulzer, 2019-05-30:
Glad SPMe works :) What's the full error you're getting from the tests? Could you try:
- Running a single test file `python3 tests/unit/test_util.py` or `python3 -m tests.unit.test_util`
- Running the style tests `python3 run-tests --flake8`

Ideally, if you could look into adding an option to `.travis.yml` to run the tests on windows, we can see if it's a problem with pybamm or your local installation

martinjrobins, 2019-05-30:
I think @ferranbrosa is correct that Pybamm sticks the path to the default parameter file on top of the current directory, so none of the battery models will work unless the user is in the pybamm root folder. Think this needs to be fixed, or make this more clear to a user

martinjrobins, 2019-05-30:
I'm a bit puzzled by the error:

```
ImportError: Failed to import test module: test_util
```

@ferranbrosa, was there any other information besides this one line?

martinjrobins, 2019-05-31:
so the windows containers for travis don't support python :( https://docs.travis-ci.com/user/reference/windows/. I'm now trying to setup appveyor

martinjrobins, 2019-05-31:
TODO:
- [ ] install graphviz on appveyor
- [ ] multiple "ImportError: scikits.odes is not installed". Don't test the scikits.odes on windows? Or install sundials...?

ferranbrosa, 2019-05-31:
Regarding @tinosulzer comment: test_util runs fine (and so do some others like test_timer or test_quick_plot), but flake8 giving the following messages:

Running flake8 ...
.\run-tests.py:32:37: E231 missing whitespace after ','
.\env\Scripts\rst2html.py:14:1: E722 do not use bare 'except'
.\env\Scripts\rst2html4.py:17:1: E722 do not use bare 'except'
.\env\Scripts\rst2html5.py:24:18: E261 at least two spaces before inline comment
.\env\Scripts\rst2latex.py:14:1: E722 do not use bare 'except'
.\env\Scripts\rst2man.py:3:10: W291 trailing whitespace
.\env\Scripts\rst2man.py:18:1: E722 do not use bare 'except'
.\env\Scripts\rst2odt.py:11:1: F401 'sys' imported but unused
.\env\Scripts\rst2odt.py:15:1: E722 do not use bare 'except'
.\env\Scripts\rst2odt.py:29:5: E128 continuation line under-indented for visual indent
.\env\Scripts\rst2odt_prepstyles.py:27:1: E302 expected 2 blank lines, found 1
.\env\Scripts\rst2odt_prepstyles.py:28:1: W293 blank line contains whitespace
.\env\Scripts\rst2odt_prepstyles.py:31:1: W293 blank line contains whitespace
.\env\Scripts\rst2odt_prepstyles.py:33:59: W291 trailing whitespace
.\env\Scripts\rst2odt_prepstyles.py:34:9: E128 continuation line under-indented for visual indent
.\env\Scripts\rst2odt_prepstyles.py:34:9: E125 continuation line with same indent as next logical line
.\env\Scripts\rst2odt_prepstyles.py:38:1: W293 blank line contains whitespace
.\env\Scripts\rst2odt_prepstyles.py:41:9: E128 continuation line under-indented for visual indent
.\env\Scripts\rst2odt_prepstyles.py:42:1: W293 blank line contains whitespace
.\env\Scripts\rst2odt_prepstyles.py:48:1: W293 blank line contains whitespace
.\env\Scripts\rst2odt_prepstyles.py:57:9: F633 use of >> is invalid with print function
.\env\Scripts\rst2odt_prepstyles.py:58:9: F633 use of >> is invalid with print function
.\env\Scripts\rst2odt_prepstyles.py:63:1: E305 expected 2 blank lines after class or function definition, found 1
.\env\Scripts\rst2pseudoxml.py:14:1: E722 do not use bare 'except'
.\env\Scripts\rst2s5.py:15:1: E722 do not use bare 'except'
.\env\Scripts\rst2xetex.py:14:1: E722 do not use bare 'except'
.\env\Scripts\rst2xml.py:14:1: E722 do not use bare 'except'
.\env\Scripts\rstpep2html.py:15:1: E722 do not use bare 'except'
FAILED

ferranbrosa, 2019-05-31:
Regarding @martinjrobins comment, the full message error I get is

======================================================================
ERROR: test_util (unittest.loader._FailedTest)
----------------------------------------------------------------------
ImportError: Failed to import test module: test_util
Traceback (most recent call last):
  File "C:\Program Files (x86)\Python37-32\lib\unittest\loader.py", line 436, in _find_test_path
    module = self._get_module_from_name(name)
  File "C:\Program Files (x86)\Python37-32\lib\unittest\loader.py", line 377, in _get_module_from_name
    __import__(name)
  File "C:\Users\Planel_F\OneDrive - University of Warwick\PyBaMM\tests\unit\test_util.py", line 8, in <module>
    import pybamm
  File "C:\Users\Planel_F\OneDrive - University of Warwick\PyBaMM\pybamm\__init__.py", line 52, in <module>
    from .expression_tree.symbol import Symbol
  File "C:\Users\Planel_F\OneDrive - University of Warwick\PyBaMM\pybamm\expression_tree\symbol.py", line 6, in <module>
    import anytree
ModuleNotFoundError: No module named 'anytree'

----------------------------------------------------------------------
Ran 47 tests in 0.021s

FAILED (errors=47)

There are several instances of this error for the different modules, but they are all caused by failing to import test module. Given that the error I pasted here is for test_util which works if ran separately, I guess the issue is with `run-tests.py`.

ferranbrosa, 2019-05-31:
Also, I have been looking how to install scikits.odes in Windows and it looks quite messy. Also, I couldn't find what is the equivalent of "Python header files (python-dev/python3-dev on Debian/Ubuntu-based distributions)" for Windows. 

Is the package needed only to solve the full DFN model?

martinjrobins, 2019-05-31:
> 
> Is the package needed only to solve the full DFN model?

It is required to solve any of the DAE models, which includes the DFN model I believe.

martinjrobins, 2019-06-03:
that is great @ferranbrosa, glad you got that working. If you find a virtual machine too resource hungry, there is also Windows Subsystem for Linux, which is quite good as well. 

We'd still be keen to develop the code for native windows as well. The main sticking point that I can see is the scikits.odes dependency. They don't seem to have developed this for windows at all (there is no windows testing or installation instructions that I can see). But even if we can't get this working, this is still only an optional dependency.

PR #464 (2019-06-09): Issue 130 add windows and mac testing
# Description

Fixes #130 

Adds windows testing via appveyor and mac os x testing via travis. Both only run the unit and integration tests, without scikits.odes installed (might need a separate issue for installing this on windows/mac)

Note: I have had to skip a *lot* of tests in the case where scikits.odes are not installed. Once we have Ivan's timestepper in pybamm, might want to make this the default solver when scikits.odes are not installed

## Type of change

Please delete options that are not relevant.

- [x] New feature (non-breaking change which [... cut]
closed 2019-06-09 by martinjrobins