pybamm-team/PyBaMM #151: remove $ at start of bash lines in README.md

martinjrobins, 2019-02-15:
comment from @fcooper8472:
* The `$` at the start of each line makes it more difficult to copy and paste to the terminal.  It's marked up as a code block anyway - could they be removed?

I agree that they don't really serve a purpose, and could be removed. Any objections?

Could also be in `CONSTRIBUTING.md`, others?

tinosulzer, 2019-02-15:
Yep fine as long as it's obvious that it's something you type in the terminal. Also we probably don't need to tell people how to upgrade pip and how to uninstall pybamm?

Scottmar93, 2019-02-15:
> Also we probably don't need to tell people how to upgrade pip and how to uninstall pybamm?

I don't see the harm in adding instructions for these things. Most users can just skip over them because they know what they mean but they make the code more accessible (and potentially much less frustrating) to less experienced users.

martinjrobins, 2019-02-15:
Trouble is that instructions would depend on the system that users had already setup. For example, pip can be installed via the Ubuntu package manager (so you should probably upgrade pip via that route, definitely need sudo), or in a separate virtual environment (so get pip to update itself, plus don't need sudo), or as part of anaconda (don't know much about this). Then there are all the different mac homebrew-type package managers (again don't know much about this either). I think we should try to keep these sort of installation instructions to a minimum.

Scottmar93, 2019-02-15:
Ah okay, sounds like a can of worms. I'm for leaving that type of thing out then.

PR #154 (2019-02-18): Issue 151 dollar
# Description

Fixes #151 

Removed all `$` from README.md and CONTRIBUTING.md to make it easier to copy/past from
closed 2019-02-18 by Scottmar93