pybamm-team/PyBaMM #155: Change NumpyModelConcatenation to DomainConcatenation in FiniteVolume

Scottmar93, 2019-02-15:
As title says. This will cause errors though so need to figure out why.

tinosulzer, 2019-02-20:
Having dug into this, maybe `DomainConcatenation` isn't the most appropriate function to use here. 
`DomainConcatenation` assumes that each symbol in the list of children being passed to it has a non-empty domain, which isn't the case in `FiniteVolume.divergence` as the boundary conditions have empty domain (they are just scalars)

Scottmar93, 2019-02-25:
This issue is solved if we remove DomainConcatenation as you suggest in the discussion in #163. DomainConcatenation is very specific to the macroscale domains which I think we should avoid.

martinjrobins, 2019-02-28:
discussed this in meeting, see result here: https://github.com/tinosulzer/PyBaMM/pull/163#issuecomment-468256714

tinosulzer, 2019-03-01:
Closed by #163

closed 2019-03-01 by tinosulzer