pybamm-team/PyBaMM #161: Changable parameters

Scottmar93, 2019-02-20:
**Summary**
Add ability to have parameter values that change without recreating the tree. 

- Add an extra subclass of symbol `ChangableParameter`. 
- Do not insert any values for `ChangableParameters` when we `params.process(model)` 
- Replace each `ChangableParameter` by a reference to a vector like `y_slices` 
   - find number of changable parameters
   - assign a number, `n`, to each changable parameters 
   - replace each parameter by `changable_parameters[n]` 
- add an extra optional input to `model.concatenated_rhs.evaluate(t, y, changable_params)` which is a vector of changable parameters. 
- create a model.set_param() which takes in a dict of parameter values and puts them into the correct location in the changable_params vector

 **Motivation**
 Allows parameters to be changed easily without rebuilding the tree. Efficiently changing parameter values is a requirement of the control community. 

Time dependent functions could also be implemented within this approach. 

**Question** 
It would be possible to have this as the default approach for setting all parameters. We should consider this as it means someone doesn't need to rebuild the discretisation if the realise afterwards that they want to change a parameter value. I also don't think that this will slow thing down to any great extent. What are peoples thought on this?

martinjrobins, 2019-02-28:
after meeting, descided to 
1. When Parameter is replaced by Scalar, the Scalar node inherits the name
2. add a function to Symbol that iterates through the tree and changes all Scalar nodes to a new value

tinosulzer, 2019-03-03:
Sounds good. We will just need to be careful that scalars don't get accidentally labelled as constant by `is_constant`, in which case they might accidentally get converted to vectors and the name lost (e.g. [here](https://github.com/tinosulzer/PyBaMM/blob/9f504cdd70ba867d5b33a0095ab42f541ef81ad6/pybamm/discretisations/discretisation.py#L255)):
```python3
if new_symbol.is_constant():
      value = new_symbol.evaluate()
      return pybamm.Vector(value)
```

PR #220 (2019-03-27): Issue 161 change parameters
# Description

Basic functionality to update a parameter value without re-discretising. A Scalar object now inherits the name from Parameter
We lose out on a bit of functionality since a Scalar can no longer be assumed to be constant (and so doesn't get simplified to a Vector). So if the discretisation step isn't too slow, it might just be better to start from scratch each time. Alternatively, we could make the `simplify` function make a copy of the model so that we retain a copy of the original information for changing parameters
Fixes #161 

## Type of change

- [ [... cut]
closed 2019-03-27 by martinjrobins