pybamm-team/PyBaMM #168: Allow symbols on different domains to interact in butler-volmer

Scottmar93, 2019-02-21:
**Summary**
We need to allow symbols on different domains to multiply when we implement the butler-volmer equation for the DFN. 

We could use the surface operator as implemented in #159 to do this. When the surface operator is acted on a variable with domain=["negative particle"], the surface operator gets assigned domain=["negative electrode"]. This way the surface operator can interact with the other variable in the butler volmer. The interaction between the surface operator and the c_n would then be the only exception to the rule that symbols on different domains cannot interact. 

Thoughts @tinosulzer @rtimms ? Might be better ways of doing this.

 **Motivation**
Required to implement the DFN

Scottmar93, 2019-02-21:
Not sure if G.domain=["negative electrode"] would be a problem when discretising and using the particle boundary conditions

tinosulzer, 2019-02-21:
Seems sensible - note that Rob has changed `SurfaceConcentration` to have domain `[]` in #167 , which would also do the job

Scottmar93, 2019-02-22:
Great. Yeah that is probably a better solution as means no exceptions. Will close this issue

closed 2019-02-22 by Scottmar93