pybamm-team/PyBaMM #175: Indefinite integral

tinosulzer, 2019-02-25:
Implementing something like F(x) = int_0^x f(x) dx (a bit tricky with finite volumes, dealing with all four cases of whether the integrand / resulting integral is to be evaluated on the nodes or on the edges. 

Might be easier to implement as solving an ODE in x: dF/dx = f(x).

Scottmar93, 2019-02-27:
I think for first version we should run with just adding an extra post_process ODE as it shouldn't be too hard to implement. 

We could at a later stage just add a front end symbol for an indenfinite integral but under the hood we actually just solve the equivalent ODE.

Scottmar93, 2019-04-08:
Following discussions on the possible node/edges cases we have decided to implement indefinite integral so that it always evaluates to nodes (therefore treating it in the same manner as the variables in the model). To extract an edge value the NodeToEdge function must be used just like any other variables.

PR #265 (2019-04-10): Issue 175 indefinite integral
# Description
Implements the indefinite integral for the cases which we require (i.e. integrate current to get potential). 

Fixes #175

## Type of change

Please delete options that are not relevant.
- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tests.py --quick`.

## Further checks: 

- [x] Code is commented, p [... cut]
closed 2019-04-10 by tinosulzer