pybamm-team/PyBaMM #177: Clean up parameters

tinosulzer, 2019-02-26:
After discussions with @Scottmar93 and @rtimms we have come up with a set of parameters that is common to both lead-acid and lithium-ion models.

We now need to clean up the files `standard_parameters` and `standard_parameters_lead_acid` to use this common parameter framework. We should also use more informative names in the parameter strings as there is no reason to use an abbreviation there, e.g. 
```python3
c_max = Parameter("maximum electrolyte concentration")
```

tinosulzer, 2019-02-27:
Some questions as I'm going along:
- Whether to use the module as a class, as we're currently doing, or use explicit classes
- How much to write in the API docs when the Parameter strings are already explicit (but combined parameters, e.g. dimensionless, don't get those explicit string names)

Scottmar93, 2019-02-27:
I quite liked it being a module because we can just do:

```from pybamm.standard_parameters import ln, lp, gamma_n, etc```

if we use an explicit class then we either need to keep doing class.parameter which make the equations harder to read or write a big list of a=class.a etc which defeats the point.

I'm not sure on the API documentation. Is the goal of API documentation to only display the things needed to interact with the module (in which case we should just show the dimensionless parameters and hide the dimensional ones) or should it offer a complete description of everything within the function?

Scottmar93, 2019-02-28:
We should also convert the submodels into the set of dimensionless parameters we agree upon. This might be a separate issue.

tinosulzer, 2019-03-03:
@rtimms and/or @Scottmar93 could you take a look at this branch please and make sure I haven't messed up the lithium-ion parameters and models. Feel free to just pull the branch locally and make your own edits. See also comments in the code on some things I wasn't sure about

Note that I couldn't get the imports to work as suggested by Scott, which means it's still a bit messy in the submodels

Scottmar93, 2019-03-03:
Happy to pick this up after I finish the 1vmany particles issue

tinosulzer, 2019-03-06:
Making a PR for this, specific parameters can be updated within model issues

PR #188 (2019-03-07): Issue 177 clean up parameters
# Description

- Change parameter notation to be consistent between different chemistries
- Change parameter names to be more informative
-  Add some functions (diffusivity, etc)
- Make parameters take up less space in model statement

Fixes #177 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is commented, parti [... cut]
closed 2019-03-07 by tinosulzer