pybamm-team/PyBaMM #185: Implement cut-off events in solvers

tinosulzer, 2019-03-02:
Solver needs to stop when certain events occur such as concentration going below zero or voltage goes below a threshold.
These events could be implemented in the model, e.g. 
```python3
self.events = [c_e, V-V_threshold]
```
and then the solver knows to stop if any of the values in `self.events` go negative.

**Motivation**
Solutions with concentration below zero are unphysical. In particular, for lead-acid, the open-circuit voltage is a logarithmic function of concentration, so we get a run-time warning if the concentration goes below zero, e.g. [here](https://travis-ci.org/tinosulzer/PyBaMM/jobs/500904077#L2387)

tinosulzer, 2019-03-04:
Have implemented without problems for `scikits.odes`, but running into an issue with Scipy. It works fine if there is only one termination event, but when trying to combine events with a concatenation it raises the error:
```
TypeError: only size-1 arrays can be converted to Python scalars
```
See e.g. [here](https://travis-ci.org/tinosulzer/PyBaMM/jobs/501765618#L2471). The relevant test is [here](https://github.com/tinosulzer/PyBaMM/blob/3eaa34dd7f7029ab209f4d719dab6bdf75ab2615/tests/test_solvers/test_scipy_solver.py#L38)

I can't isolate this error in debug mode so am a bit stuck. Any ideas @martinjrobins @fcooper8472 @Scottmar93 @rtimms ?

rtimms, 2019-03-06:
I guess a hacky way around it is to just take the product of the events, but may be an issue if two go negative together... For Scipy I think you need to write each event as a function and then pass the concatenated functions, e.g.

```
def event1(t, y):
    return y - 3

def event2(t, y):
    return t - 3

events = [event1, event2]
```

Then give `events` to Scipy

tinosulzer, 2019-03-06:
That works, thanks!

PR #187 (2019-03-07): Issue 185 cutoff event
# Description

Implement cut-off events in solvers, add cut-off events to models.
Fixes #185 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] Tests added that prove fix is effective or that feature works
- [ ] Any dependent changes have been merged and publishe [... cut]
closed 2019-03-07 by martinjrobins