pybamm-team/PyBaMM #189: Symbolic differentiation

tinosulzer, 2019-03-06:
We need a function (e.g. `pybamm.diff`) that calculates the derivative of an expression tree with respect to a variable. For example, some lead-acid models require the user to provide a function for the open-circuit potential `U(c)` while others require its derivative `dU/dc` (and also for the exchange-current density). 

We could either ask the user to provide the appropriate derivatives manually at the same time as the functions themselves, or calculate the derivative using the expression tree and appropriate applications of the chain rule and power rule. I think the latter is safer

tinosulzer, 2019-03-07:
Autograd is working well for the arbitrary numpy functions, but the drawback is that we have to import `autograd.numpy` instead of `numpy`. Are we happy to force the user to do this?

rtimms, 2019-03-08:
It looks like `autograd` is probably the easiest tool out there for calculate the Jacobian for use in the solver, which would force us to use `autograd.numpy` anyway. Maybe there are other options, but from what I can see this is probably the simplest fix

martinjrobins, 2019-03-08:
does that mean that the user *wouldn't* be able to use normal numpy?

tinosulzer, 2019-03-11:
We could do something like 
```python3
try:
    import autograd.numpy as np
except ModuleNotFoundError:
    import numpy as np
```

Then throw errors/warnings that the functionality provided by autograd (symbolic differentiation, automatic jacobian in #197) can't be used if it's not installed.
Might be best to do this in `__init__.py` and then `pybamm.numpy as np` whenever needed so that we don't need to repeat the above 4 lines every time.

In any case, autograd is easy to install with pip and seems well maintained so it should be fine to just include in `setup.py`

martinjrobins, 2019-03-11:
seems sensible to start out using autograd for both this issue and #197 and we can see how the PR looks. if we use autograd this should be fairly straightforward to implement

tinosulzer, 2019-03-12:
I've submitted a PR for this issue

tinosulzer, 2019-03-13:
Re-opening: need to deal with `FunctionParameter` nodes better

PR #196 (2019-03-12): Issue 189 symbolic differentiation
# Description

Implement symbolic differentiation using simple calculus rules and autograd for the `Function` objects.
Requires implementing functions using `autograd.numpy` instead of `numpy`.
Fixes #189 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [... cut]
PR #202 (2019-03-20): Issue 189 diff function parameter
# Description

Implement `diff` for a `FunctionParameter` object. The `value` attribute of a `Scalar` has to be converted to a float to work with autograd but this doesn't seem to cause any problems
Fixes #189 

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] [... cut]
closed 2019-03-20 by tinosulzer