pybamm-team/PyBaMM #199: fill out example notebooks

martinjrobins, 2019-03-13:
This is a parent issue, see individual issues below for details...

The following notebooks need to be filled out 
- [ ] Run a pre-defined model
- [ ] Compare models
- [ ] Add a model
- [ ] Change the settings (parameters, discretisation or solver)
- [ ] Add parameters
- [ ] Add a discretisation
- [ ] Add a solver
- [ ] Expression tree structure
  - [ ] Overview
  - [ ] Expression tree
  - [ ] Setting parameters
  - [ ] Discretising a model
  - [ ] Solving a model
- [ ] Lithium-ion models
  - [ ] Single-Particle Model
  - [ ] Single-Particle Model with electrolyte
  - [ ] Doyle-Fuller-Newman Model
- [ ] Lead-acid models
   - [ ] Full porous-electrode
   - [ ] Leading-Order Quasi-Static
   - [ ] First-Order Quasi-Static
   - [ ] Composite
- [ ] Finite Volumes
- [ ] Inbuilt SciPy solver
- [ ] scikits.odes solvers

tinosulzer, 2019-03-13:
You mentioned some of these would be better as not notebooks (tutorials in the API docs?) - can we decide in this issue which examples should be notebooks and which should be something else?

martinjrobins, 2019-03-13:
Yes, absolutely. Feel free to go through and edit this list.

I had a few ideas for an expression tree notebook, which is why I've started this issue (and #200). But I hadn't fully decided what notebooks we need and which ones would be better in the api docs

rtimms, 2019-05-15:
looks like these are all done now. are there any extra examples we need at this moment in time @tinosulzer @Scottmar93 @martinjrobins ?

tinosulzer, 2019-05-15:
Nope, happy to close this issue

closed 2019-05-17 by martinjrobins