pybamm-team/PyBaMM #204: consistent initialisation of DAE systems

tinosulzer, 2019-03-20:
For spatially-dependent algebraic equations it's impossible for the user to manually specify consistent initial conditions. 
There needs to be a function (after discretisation and before solver) that takes the algebraic model and the initial conditions from the differential models and computes consistent initial conditions for the algebraic model using those. Ideally we would never ask the user for initial conditons for the algebraic systems
I don't know whether scikits has this inbuilt. From a brief search through the documentation, [[ida.pyx](https://github.com/bmcage/odes/blob/master/scikits/odes/sundials/ida.pyx#L740)] this suggests IDA can compute either `y0` given `ydot0` or `ydot0` given `y0`

@martinjrobins any experience with this?

tinosulzer, 2019-03-20:
Possible workflow:
1. User provides initial conditions `y0_diff` for the differential equations and guesses for the initial conditions `y0_alg_guess` for the algebraic equations
2. Process model as usual
3. Solve `model.concatenated_algebraic(0, [y0_diff, y]) = 0` for `y` with `y0_alg_guess` as the initial guess, e.g. using `scipy.optimize.fsolve`
3a. Find a way to calculate `ydot0`?
4. Feed the result into the DAE solver and continue

**or**

3. Run the ode solver treating `model.concatenated_algebraic(t, [y0_diff, y])` as system of ODEs in `t` and `y`, with initial conditions `y0_alg_guess`, and hope that we reach a steady state that we can use for step 4

tinosulzer, 2019-03-20:
Finding consistent initial conditions is working well for simple tests. I'm now wondering whether we need `ydot0` or whether we can always set that to 0 by default

martinjrobins, 2019-03-20:
If IDA can calculate a consistent ydot0 from y0, that seems to be the best way forward

PR #208 (2019-03-20): Issue 204 dae initialisation
# Description

Calculate initial conditions for the algebraic equations. User must still provide an initial guess to start the root finding algorithm
Fixes #204 

## Type of change

- [x] New feature (non-breaking change which adds functionality)
- [x] Breaking change (fix or feature that would cause existing functionality to not work as expected)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [ ] Code is  [... cut]
closed 2019-03-20 by martinjrobins