pybamm-team/PyBaMM #209: example notebook: Run a pre-defined model

martinjrobins, 2019-03-22:
see #199

martinjrobins, 2019-03-28:
@tinosulzer @Scottmar93 @rtimms 

Would help here if I knew what sort of outputs/visualisations would be standard for battery modelling?

tinosulzer, 2019-03-28:
The main ones are voltage (and current if it's interesting i.e. not just constant, temperature if it's in the model) as a function of time.
Then usually other spatially-dependent variables (concentration, interfacial current) as functions of x (or particle concentration as a function of r in the SPM/SPMe) at fixed t. A slider plot where you can vary t would be ideal for the notebooks.
I don't know the standard way of plotting particle concentration when it varies in x, r and t.

rtimms, 2019-03-28:
for particle conc. can do a 2D contour plot in r and x, with a slider for time (or for a fixed time). some people just plot the surface concentration as a function of time, with a separate line for each particle on the same graph, but this can get messy!

Scottmar93, 2019-03-28:
Also averaged quantities against time. This is useful for comparing SPMe with the averaged quantities from DFN.

PR #259 (2019-04-08): Issue 209 example model
# Description

added an example notebook for the SPM model

Fixes #209 

## Type of change

Please delete options that are not relevant.

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tests.py --quick`.
closed 2019-04-08 by martinjrobins