pybamm-team/PyBaMM #227: Post-process model variables

tinosulzer, 2019-03-28:
**Summary**
 After solving a model, post-process the variables to objects that can be evaluated as functions of `x` and `t`, rather than `t` and `y`, and without having to loop through indices.
For example, a new class `ProcessedVariable` that has an attribute `entries` which is the variables evaluated at all the discrete values of `t` and `y` from the solver, and a method `evaluate(x, t)` that interpolates appropriately.

 **Motivation**
Easier plotting and comparison of variables, e.g. as suggested by @martinjrobins in #209 . Will also help #69 for example.

martinjrobins, 2019-03-28:
This would be good. Could also allow `x` or `t` to be vectors or scalars so you could get the value at a particular x over all time points, or the value at a particular t over all space points

tinosulzer, 2019-03-29:
Should `ProcessedVariable` be part of the expression tree (inherit from `Symbol`) or be something completely separate?

martinjrobins, 2019-03-29:
I think separate. It's `evaluate` has a different meaning to `Symbol`, so I don't think it can inherit from it

tinosulzer, 2019-04-03:
It might make sense to overload the `__call__` function for the `ProcessedVariable` class, instead of using `.evaluate()`. Thoughts @martinjrobins ?

martinjrobins, 2019-04-04:
yea, I'd be happy with either really. `evaluate` has the advantage that its consistent with the expression tree, but I think the __call__ looks nicer for user-facing code. I'd probably lean towards the __call__.

tinosulzer, 2019-04-04:
Cool, seems useful as a way to highlight that this is a different `evaluate` to the tree evaluate as well

PR #235 (2019-04-01): Issue 227 post process
# Description

Post process a variable for easier evaluation, with interpolation. Currently won't work for:
- Symbols that are like grad(u) that get evaluated on edges rather than nodes
- Microscale symbols

To do:
- [x] implement this in the model tests 

Fixes #227 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is co [... cut]
PR #255 (2019-04-04): #227 change evaluate to __call__
# Description

Use `.__call__` instead of `.evaluate` to evaluate a processed variable
Fixes #227 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] Tests added that prove fix is effective or that feature works
- [x] Any dependent changes have been merg [... cut]
closed 2019-04-04 by tinosulzer