pybamm-team/PyBaMM #237: SPM model only solved at two time points

martinjrobins, 2019-04-01:
**Describe the bug**

When solving the SPM model, whatever the time points put into the solver, the solver only returns the solution at two time points. This is not caught by the current tests

**To Reproduce**
Steps to reproduce the behaviour:
1. Run the `test_li_ion.py` test
2. print out the time vector `T` in `test_surface_concentration`

**Expected behaviour**

solver should return the solution at all the time points specified (np.linspace(0,1,100))

tinosulzer, 2019-04-01:
It looks like the concentration in the negative electrode is hitting the cut-off after the second time-step. 
Running with `modeltest.test_all(t_eval=np.linspace(0, 0.01, 100))`, or setting a higher initial concentration with
```python3
model.default_parameter_values.update({"Initial concentration in negative electrode": 6000})
```
gives the solution at more time steps. 
There must be a typo in the parameters (diffusivity too low, initial conditions too low or boundary conditions too high?), because the way the model is non-dimensionalised you would expect the solution to run for almost one unit of time - can you check this @Scottmar93 @rtimms ?

Scottmar93, 2019-04-01:
In process of getting DFN to work I found that all timescale ratios were entered the wrong way round and there were some errors in the way the ocps were entered. I have corrected these along with some other small errors within in the branch issue-74-implement-DFN. You could merge in `pybamm/parameters/standard_parameters_lithium_ion.py`, `input/parameters/lithium-ion/graphite_mcmb2528_ocp_Dualfoil.py`, and `lico2_ocp_Dualfoil.py`. Although this branch isn't stable yet. Hopefully should be ready by tomorrow somepoint.

Scottmar93, 2019-04-02:
may in addition be problems related to #239

Scottmar93, 2019-04-02:
made pull request #241 which is stable and addresses most of the parameters problems (except #239)

Scottmar93, 2019-04-03:
@martinjrobins Is this issue now resolved for you?

martinjrobins, 2019-04-04:
yup, this looks good, thanks @Scottmar93!

closed 2019-04-04 by martinjrobins