pybamm-team/PyBaMM #239: Parameter import errors

Scottmar93, 2019-04-02:
**Describe the bug**
Importing the parameters in the current form leads to changing `standard_parameters` when they are imported later. This leads to weird behaviour such as tests passing or failing dependent upon the order in which they are run. 

**To Reproduce**
Enter
```
in[1]: param = pybamm.standard_parameters
in[2]: param.__dict__.update(pybamm.standard_parameters_lithium_ion.__dict__)
```
(this is how we currently import parameters at the top of each model)
then enter:
in[1]: pybamm.standard_parameters
out[1]: <module 'pybamm.parameters.standard_parameters_lithium_ion' from '/home/user/PycharmProjects/PyBaMM/pybamm/parameters/standard_parameters_lithium_ion.py'>
```

**Expected behaviour**
We should instead have:
```
in[1]: pybamm.standard_parameters
out[1] <module 'pybamm.parameters.standard_parameters' from '/home/user/PycharmProjects/PyBaMM/pybamm/parameters/standard_parameters.py'>
```
otherwise `pybamm.standard_parameters` depends on previous actions.

tinosulzer, 2019-04-02:
Yeah there must be a better way of doing it. One option would be to make the `standard_parameters.py` explicit classes (rather than module classes) and import from those, then we could use `super().__init__` in `standard_parameters_lithium_ion` etc. However, this would mean we'd have loads of "unnecessary" `self.`  floating around. I couldn't figure out a good middle ground when I was doing this so stuck with the hack

martinjrobins, 2019-04-03:
you could keep the parameters separate when you are referencing them in a model? e.g.

```python
param = pybamm.standard_parameters
param_lion = pybamm.standard_parameters_lithium_ion
```
or perhaps better done as:

```python
import pybamm.parameters.standard_parameters as param
import pybamm.parameters.standard_parameters_lithium_ion as param_lion
```

Scottmar93, 2019-04-03:
Might also be okay to remove the standard parameters file and simply have all parameters in both standard_parameters_lithium_ion and standard_parameters_lead_acid. I think it also makes it a little easier to read and work with as you don't have to think is this a parameter specific to lithium ion or worry that its also a parameter for lead acid but defined slightly differently in lead acid.

Scottmar93, 2019-04-03:
Going to implement separate parameter files.

PR #246 (2019-04-03): Issue 239 seperate parameters
# Description

Removed the standard parameters file and put majority of parameters into the standard_parameters_lithium_ion and standard_parameters_lead_acid files. I have created 2 small parameters files: geometric_parameters and electrical_parameters for thing which need chemistry independent parameters like creating a geometry and the current function in homogeneous reactions.

All uses of the .__dict__.update have been removed and tests fixed where parameter leak was occurring.

Fixes #239 

## Type of change

Please delete options that are not relevant.

- [x [... cut]
closed 2019-04-03 by Scottmar93