pybamm-team/PyBaMM #250: How should we implement dimensional outputs?

Scottmar93, 2019-04-04:
Does anyone have a good idea how to do this? It would be good to discuss on Monday. 

A couple of things to note: 
- we can either output a separate dictionary with dimensional output variables or somehow include dimensional variables into the current output dictionary (avoiding any name clashes with a "Dimensionless... at start of variable names).
- If the dimensional output variables are entered within the submodels then we would need to be careful that the scalings used are consistent across chemistries. 
- we could maybe add a dimensionalise function for each chemistry which searches through variables and finds the correct scalings to apply (This may be more complex than its worth though). 

 **Motivation**
Dimensional variables are needed in practice.

martinjrobins, 2019-04-08:
wouldn't the dimensional outputs go in `model.variables` along with all the other needed output variables?

Scottmar93, 2019-04-08:
Following meeting we will insert dimensional variables into the `model.variables` dictionary and include the units of the variable in the name.

PR #262 (2019-04-08): Add dimensional outputs to SPM, SPMe, DFN
# Description
As in title

Fixes #250 

## Type of change

Please delete options that are not relevant.

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ py [... cut]
closed 2019-04-08 by tinosulzer