pybamm-team/PyBaMM #253: Simplify Concatenation (and subsequent bug)

tinosulzer, 2019-04-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

martinjrobins, 2019-04-08:
@tinosulzer, do you mean the original tree was

```
Matrix (9,8)   ( vector (8,) @ Matrix(8,9) )
```

But `vector (8,) @ Matrix(8,9)` gives `vector(9,)`?

tinosulzer, 2019-04-08:
Sorry that wasn't clear, there is a mix of matrix multiplications and elementwise multiplications:
original:
```
Matrix (9,8) @ (vector (8,) * Matrix(8,9))  ---> Matrix(9,8) @ Matrix(8,9) ---> Matrix(9,9)
```
after simplification:
```
(Matrix (9,8) @ vector (8,)) * Matrix(8,9)  ---> Vector(9,) * Matrix(8,9) ---> Error
```

martinjrobins, 2019-04-08:
but `vector (8,)` is a row vector, so  `vector (8,) * Matrix(8,9)`  shouldn't work? should be `vector (8,1)`?

```
In [1]: import numpy as np

In [2]: v = np.ones(8)

In [3]: m = np.ones((8,9))

In [4]: v*m
---------------------------------------------------------------------------
ValueError    Traceback (most recent call last) <ipython-input-4-c1d8335750dc> in <module>
----> 1 v*m

ValueError: operands could not be broadcast together with shapes (8,) (8,9)
```

martinjrobins, 2019-04-08:
so if you were to expand out the brackets in `m2@(v*m)`, what would you get mathematically? I'm not sure..

tinosulzer, 2019-04-08:
It looks like the `*` operator does matrix multiplication by default for scipy matrices, so we should use `A.multiply(B)` instead:
```
In [13]: sparse.coo.coo_matrix((8,9))*sparse.coo.coo_matrix((8,9))
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-13-c73b5d00c9d5> in <module>()
----> 1 sparse.coo.coo_matrix((8,9))*sparse.coo.coo_matrix((8,9))

~/.local/lib/python3.6/site-packages/scipy/sparse/base.py in __mul__(self, other)
    476         if issparse(other):
    477             if self.shape[1] != other.shape[0]:
--> 478                 raise ValueError('dimension mismatch')
    479             return self._mul_sparse_matrix(other)
    480 

ValueError: dimension mismatch

In [14]: sparse.coo.coo_matrix((8,9))*sparse.coo.coo_matrix((9,8))
Out[14]: 
<8x8 sparse matrix of type '<class 'numpy.float64'>'
	with 0 stored elements in Compressed Sparse Row format>

In [15]: sparse.coo.coo_matrix((8,9))@sparse.coo.coo_matrix((9,8))
Out[15]: 
<8x8 sparse matrix of type '<class 'numpy.float64'>'
	with 0 stored elements in Compressed Sparse Row format>

In [16]: sparse.coo.coo_matrix((8,9))@sparse.coo.coo_matrix((8,9))
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-16-ce8641f5818e> in <module>()
----> 1 sparse.coo.coo_matrix((8,9))@sparse.coo.coo_matrix((8,9))

~/.local/lib/python3.6/site-packages/scipy/sparse/base.py in __matmul__(self, other)
    557             raise ValueError("Scalar operands are not allowed, "
    558                              "use '*' instead")
--> 559         return self.__mul__(other)
    560 
    561     def __rmatmul__(self, other):

~/.local/lib/python3.6/site-packages/scipy/sparse/base.py in __mul__(self, other)
    476         if issparse(other):
    477             if self.shape[1] != other.shape[0]:
--> 478                 raise ValueError('dimension mismatch')
    479             return self._mul_sparse_matrix(other)
    480 

ValueError: dimension mismatch

In [17]: sparse.coo.coo_matrix((8,9)).multiply(sparse.coo.coo_matrix((8,9)))
Out[17]: 
<8x9 sparse matrix of type '<class 'numpy.float64'>'
	with 0 stored elements in Compressed Sparse Row format>
```

martinjrobins, 2019-04-08:
ok. But this doesn't relate to this bug? probably better to open that as a separate issue

tinosulzer, 2019-04-08:
> so if you were to expand out the brackets in `m2@(v*m)`, what would you get mathematically? I'm not sure..

It looks like this isn't possible in general

> ok. But this doesn't relate to this bug? probably better to open that as a separate issue

Ok will open separate issue and hope nothing breaks ...  We might be ok because the original case should actually be
```
div(D(c)*grad(c)) 
---> Matrix (9,8) @ (vector (8,) * (Matrix(8,9) @ vector(9,)))  
---> Matrix (9,8) @ (vector (8,) * vector(8,))
---> Matrix(9,8) @ vector(8,) 
---> vector(9,)
```

martinjrobins, 2019-04-08:
I've fixed the bug so that `m2@(v*m).simplify()` will not try to expand the brackets around the *, but still getting dimension errors for the model tests. I've had a look at the model equations, but they are too long to see what the problem is. How did you manage to find the problem earlier @tinosulzer, any tips?

tinosulzer, 2019-04-08:
Try putting a try-except block around the operation that is giving a dimension error, then you can have a look at what the children evaluate to and why it's giving an error

martinjrobins, 2019-04-08:
ok, think I might have figured out the problem, when I fixed the issue you described above with the sparse multiplication the error message became much more meaningful. The statevector is being put in during the solve as a row vector, rather than a column vector, so at some point there is something like:

`vector(99,) * sparse_matrix(99,100)`

this was getting treated like a matrix multiplication, which is fine, but then I fixed it so that it was treated as a broadcasting scalar multiply, it started complaining the dimensions didn't agree, which is also right, it should be

`vector(99,1) * sparse_matrix(99,100)`

Does that sound right?

tinosulzer, 2019-04-08:
That sounds right but even with the sparse multiplication fixed there are no errors like this on the non-simplified model, and I am guessing that this is because there are never any vector-matrix multiplications, only matrix-vector (e.g. grad(c) or div(grad(c)) or vector-vector (e.g. c*grad(c))

martinjrobins, 2019-04-08:
I think the problem is that there are things like:
```
vector(99,) * ( sparse_matrix(99,100) @ statevector(99,) )
```

if you just evaluate that, you do the inner matrix-vector multiply which gives you a vector, then multiply that by the outer vector. All good. But if you try to simplify by doing the `vector(99,) * sparse_matrix(99,100)` to give you a `sparse_matrix(99,100)`,  then you run into problems

I've pushed a hacky fix to `pybamm.Multiplication` which takes into account 1d numpy vectors, and a few other misc fixes that arose after that one. All the tests pass now, which is good. Is the original bug you found in this issue fixed?

tinosulzer, 2019-04-08:
Have added (but not tested) simplification of `DomainConcatenation`, which required removing the line in `symbol.simplify` that removed domain (`DomainConcatenation` needs to know the domains).
Good news is it runs and simplifies the tree, bad news is we don't get much speed gain :confounded: 

In `dae_solver.py`:
```
In [2]: %timeit concatenated_rhs.evaluate(0,y0)                                                                                                                                                             
1.72 ms ± 14.4 µs per loop (mean ± std. dev. of 7 runs, 1000 loops each)

In [3]: %timeit model.concatenated_rhs.evaluate(0,y0)                                                                                                                                                       
1.93 ms ± 85.9 µs per loop (mean ± std. dev. of 7 runs, 1000 loops each)
```

tinosulzer, 2019-04-08:
It could be the reshapes in `Multiplication` slowing things down in the simplified tree (more reshapes needing to happen there than in the original tree)

martinjrobins, 2019-04-08:
that is a shame :( I've been doing the simplify for domain concatentation as well, without the domain information, this should pass all the tests now.

PR #263 (2019-04-08): Issue 253 simplify bug
# Description

Fixes #253 

See issue for description

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tests.py --quick`.

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] Tests added that prove fix is effective or that fe [... cut]
closed 2019-04-08 by tinosulzer