pybamm-team/PyBaMM #267: Eliminate repeated operations

tinosulzer, 2019-04-10:
When the same operation needs to be performed in several parts of the model, we currently perform it from scratch each time, even on the simplified tree. One that appears very often is the interfacial current density, `j=j0(c)*sinh(phi_s - phi-U(c))` (which is expensive to calculate).

Instead, we could keep a dictionary
```python3
{symbol.id: result of being evaluated}
```
that gets passed to evaluate; if the symbol id has already been evaluated, then evaluate just needs to read that dictionary to get the value, rather than having to repeat the calculation.

tinosulzer, 2019-04-10:
Update: @Scottmar93 has suggested keeping an attribute (`evaluated_value`?) that we can look up to see if we have already evaluated the symbol at that `t` and `y`, and update otherwise

martinjrobins, 2019-04-10:
The dictionary sounds like a good idea. Another option to consider is to construct a execution graph from the tree, you would need to identify the common subtrees in the expression tree, and combine them as a single subtree with 2 parents, hence converting your tree (max 1 parent) to a graph (any number of parents).

You would need a `Graph` class, which holds a list of parents, a list of children, and a reference to the node it will execute, as well as an `execute` function of its own

tinosulzer, 2019-04-11:
I don't really understand the execution graph so am implementing with dictionary
One potential issue is that the known_evals dict could in theory be providing a value that is different to the actual value of the node. Do we want to do something to counter this (other than writing good tests)? e.g. a random check every 100th operation that bypasses known_evals and makes sure `known_evals[node.id] = node.evaluate(t,y)`?

martinjrobins, 2019-04-11:
we should definitely test this thoroughly (perhaps we could test every model in pybamm during the tests), but I don't think the random check at run-time is necessary.

tinosulzer, 2019-04-11:
Ok, have added a test to all the models to make sure that the result using simplify and/or known_evals is the same as without

PR #269 (2019-04-11): Issue 267 repeated ops
# Description

Store known results of operations using `known_evals` dictionary. Introduces some (very minor) overhead to the models that are already quick, but greatly speeds up the models that are slow.
Fixes #267 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
-  [... cut]
closed 2019-04-11 by martinjrobins