pybamm-team/PyBaMM #271: node to edge and boundary value functions

tinosulzer, 2019-04-11:
Use matrix operations instead of functions to implement these. 

In particular, we should be able to define a single matrix that calculates a boundary value and concatenates it to the existing state vector

rtimms, 2019-04-11:
This will help with speeding up jacobian in #197 as autograd has an `elementwise_grad` which gives the diagonal of the jacobian *if* the jacobian is diagonal, e.g.

```
In [29]: x = 2 * np.ones(3)                                                                                                                                                                                 

In [30]: def f(x): 
    ...:     return x ** 2 
    ...:                                                                                                                                                                                                    

In [31]: autograd.jacobian(f)(x)                                                                                                                                                                            
Out[31]: 
array([[4., 0., 0.],
       [0., 4., 0.],
       [0., 0., 4.]])

In [32]: autograd.elementwise_grad(f)(x)                                                                                                                                                                    
Out[32]: array([4., 4., 4.])
```
If we know all our functions behave this way, then we can just make a sparse diag as the result

tinosulzer, 2019-04-11:
Great - do you fancy giving it a go?

rtimms, 2019-04-11:
Yep, I’m not here tomorrow, but can have a go next week!

rtimms, 2019-04-17:
I've changed `compute_diffusivity` to be a matrix multiplication, but now we can end up with matrix multiplication in the denominator e.g. in `MacInnesStefanMaxwell` the term `pybamm.grad(c_e) / c_e` now goes to `pybamm.grad(c_e) / (M @ c_e)`. This is of the right size, but matrix multiplication in the denominator gives an error in `simplify_multiplication_division`. I'm having a little trouble understanding exactly how this is working. @tinosulzer @martinjrobins would you be able to help please?

tinosulzer, 2019-04-17:
Would it work if you changed `pybamm.grad(c_e) / c_e`  to `pybamm.grad(log(c_e))` as a workaround ?

tinosulzer, 2019-04-17:
^ apparently that doesn't work.

@martinjrobins was there any particular reason why we couldn't have a matrix multiplication in the denominator?

martinjrobins, 2019-04-17:
I assumed that there shouldn't be any matrices / vectors on the
denominator. Is this not the case?

tinosulzer, 2019-04-17:
If we use matrix-multiplication instead of a `Function` for calculating diffusivities (node to edge), then a matrix multiplication appears in the denominator:
```
disc.process_symbol(grad(c) / c) ---> (M1 @ c) / (M2 @ c)
```
where both `M1` and `M2` have size (n-1, n) (`n=len(c)`)

We could also just rewrite the model: `1/c * grad(c)` instead of `grad(c)/c` - @rtimms  can you see if that works?

martinjrobins, 2019-04-17:
ok. So I guess the divide in this case is an element-wise divide, so this is valid? Perhaps I should allow mat-muls on the denominator then?

tinosulzer, 2019-04-17:
yeah, in this case it's elementwise, so this kind of mat-mul can be allowed

rtimms, 2019-04-17:
yep, changing to `1/c * grad(c)` works! still some other errors to with shapes on other tests which I'll investigate tomorrow

rtimms, 2019-04-18:
@martinjrobins  @tinosulzer this has caused some more issues to come up with simplifying mat-muls, e.g. 

```
ipdb> model.concatenated_rhs.children[4].children[1].children[0].children[0].children[0].children[0].children[0]                                                                                            
MatrixMultiplication(-0x787c479f73fec89a, @, children=['Sparse Matrix (100, 99)', 'Sparse Matrix (99, 100) @ -domain concatenation ** Bruggeman coefficient * Sparse Matrix (99, 100) @ domain concatenation'], domain=['negative electrode', 'separator', 'positive electrode'])
ipdb> model.concatenated_rhs.children[4].children[1].children[0].children[0].children[0].children[0].children[0].simplify()                                                                                 
MatrixMultiplication(-0x143222ae39357139, @, children=['Sparse Matrix (100, 100)', '-y[104:204] ** 1.5 * Sparse Matrix (99, 100) @ y[4:104]'], domain=[])
ipdb> model.concatenated_rhs.children[4].children[1].children[0].children[0].children[0].children[0].children[0].evaluate(0,y0)                                                                             
array([0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.])
ipdb> model.concatenated_rhs.children[4].children[1].children[0].children[0].children[0].children[0].children[0].simplify().evaluate(0,y0)                                                                  
*** ValueError: operands could not be broadcast together with shapes (100,1) (99,1)
```

here something like `A @ ((B @ u) * (C @ v))` is getting simplified to `(A @ B) @ (u * (C@v))`

martinjrobins, 2019-04-18:
`A @ ((B @ u) * (C @ v))` shouldn't be simplified due to the `*` . I can write a test for this, are u and v StateVectors?

rtimms, 2019-04-18:
Yup, u and v are StateVectors. Thanks!

martinjrobins, 2019-04-18:
this is now fixed in PR #302

tinosulzer, 2019-04-23:
how is this going @rtimms ?

rtimms, 2019-04-23:
@tinosulzer have just pushed the matmul version of boundary value and node to edge. they appear to work apart from a few shape issues (see #301). hopefully this should be OK after that is fixed.

PR #316 (2019-04-27): Issue 271 node to edge
# Description

`compute_diffusivity`  and `boundary_value` are now matmuls in `finite_volume`

Fixes #271 

## Type of change

Please delete options that are not relevant.

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make  [... cut]
closed 2019-04-27 by rtimms