pybamm-team/PyBaMM #285: add manual DAE integrator

martinjrobins, 2019-04-16:


martinjrobins, 2019-06-23:
Hi @ikorotkin, I've wrapped up dae-cpp using pybind11, and created an example of how to use the wrapper in the test file `tests/unit/test_solvers/test_daecpp_solver.py`, based on your robertson example

Have a look and see if you can get it to work. After that, the remaining thing to do is to write a new solver within the pybamm framework that uses the wrapper.

dae-cpp and pybind11 are included as submodules in pybamm, so you will need to run these commands to get the code (this is after you clone the pybamm repo)

```bash
git submodule init
git submodule update
```

If you have already installed pybamm, you will need to reinstall it using pip so that the dae-cpp gets compiled. ie. something like (in the root pybamm dir)

```bash
pip uninstall pybamm
pip install -e .
```

You will of course need all the dependencies that you would ordinarily need to compile dae-cpp

after that you should be able to run the test for the wrapper:

```bash
python tests/unit/test_solvers/test_daecpp_solver.py
```

I had a problem where MKL could not load the correct libraries, it would give this error at run-time:
```
Intel MKL FATAL ERROR: Cannot load libmkl_avx.so or libmkl_def.so.
```

I found that I could get rid of this by preloading the mkl libraries, using:

```
export LD_PRELOAD=/opt/intel/mkl/lib/intel64/libmkl_def.so:/opt/intel/mkl/lib/intel64/libmkl_avx2.so:/opt/intel/mkl/lib/intel64/libmkl_core.so:/opt/intel/mkl/lib/intel64/libmkl_intel_lp64.so:/opt/intel/mkl/lib/intel64/libmkl_intel_thread.so:/opt/intel/lib/intel64_lin/libiomp5.so
```

This is clearly not ideal, so if you can shed any light on this it would be great. It could be something to do with how I am compiling dae-cpp and the wrapper, you can see what is done in the `CMakeLists.txt` file in the root pybamm directory

ikorotkin, 2019-06-23:
Thank you @martinjrobins, that looks great! I'll start looking into this tomorrow.

martinjrobins, 2019-07-08:
I've wrapped the stop_condition for dae-cpp RHS class now. Once you have defined a stop condition function in python:

```python
def stop(x, t):
   return some_function_of(x, t)
```

then you can set this in the rhs class using the `set_stop_condtion` method:

```python
rhs = pydae.RHS(fun_rhs)
rhs.set_stop_condition(stop)
```

tinosulzer, 2021-03-29:
Not something we are planning on doing at the moment

closed 2021-03-29 by tinosulzer