pybamm-team/PyBaMM #292: example notebook/tutorial: solving model & adding a solver

tinosulzer, 2019-04-17:
Similar to #238

tinosulzer, 2019-04-17:
@rtimms  would it be possible to implement mass matrix for the scikits ode solver (currently passed but not used)?

rtimms, 2019-04-18:
cvode requires the send to be given explicitly in the form y’=f(t,y). would need to invert the mass matrix to use it with solver. I’m not sure how efficient that is, or whether we should encourage use of the scipy solver or gives the odes implicitly to IDA? I think sundials recommend ARKode for odes with a nonsingular mass matrix, but this isn’t in scikits

tinosulzer, 2019-04-18:
Ok, I was thinking about the case where the mass matrix is tri-diagonal, but happy to just leave it out of the cvode solver and have it available if someone writes their own ODE solver and wants to use it

rtimms, 2019-04-18:
One option would be to check if the mass matrix is the identity and then do nothing, otherwise invert and multiply rhs by the result (maybe with a warning). There is some discussion [here](https://stackoverflow.com/questions/21693613/is-it-possible-to-compute-an-inverse-of-sparse-matrix-in-python-as-fast-as-in-ma) about inverting sparse matrices with scipy

PR #299 (2019-04-26): Issue 292 doc solver
# Description

Add a tutorial and two notebooks for solvers
Fixes #292 

## Type of change

Please delete options that are not relevant.

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tests.py --quick`.

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] Tests added that prove fix [... cut]
closed 2019-04-26 by tinosulzer