pybamm-team/PyBaMM #329: Reduce the number of deepcopies in the workflow

tinosulzer, 2019-05-02:
These are very slow (usually account for ~60% of total runtime)

- [ ] Alternative for setting parameters
- [ ] Alternative for discretising
- [ ] Alternative for simplifying
- [ ] Alternative for `orphans`

tinosulzer, 2019-05-02:
I've done this for parameter values and discretisation and it makes a difference.
Not immediately obvious how to do this for simplification. The main issue is that we do a deepcopy of the whole tree at each level. It might make more sense to have an external class that does the simplification, similarly to setting parameter values and discretising; this could get called by `Symbol.simplify()` so that this functionality remains

martinjrobins, 2019-05-02:
Yup, as per slack discussion I think this would be a good idea, so go ahead @tinosulzer

tinosulzer, 2019-05-02:
To overwrite the `orphans`, we could make our own equivalent of `deepcopy` that just iteratively creates a new object (I guess this doesn't do *quite* the same thing as `deepcopy` so we should call it something else, maybe `new_instance` or something)

PR #403 (2019-05-21): Issue 329 deepcopy
# Description

Create function `make_new_copy` to replace `deepcopy`
Fixes #329 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tests.py --quick`.

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] Tests added that prove fix is effective or that feature works
- [ [... cut]
closed 2019-05-21 by martinjrobins