pybamm-team/PyBaMM #333: Issue exporting fluxes

Scottmar93, 2019-05-02:
**Describe the bug**
Shape size error when trying to output fluxes and currents.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to SPM.ipynb
2. Run all
3. Scroll to In[17] (below the list of output variables)
4. Add the line:
```python
pybamm.ProcessedVariable(
    model.variables['Positive particle flux'], solver.t, solver.y, mesh=mesh
)
```

Get the following error message:
```python
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-17-74cbf4a13b3d> in <module>
      7 )
      8 pybamm.ProcessedVariable(
----> 9     model.variables['Positive particle flux'], solver.t, solver.y, mesh=mesh
     10 )

~/PycharmProjects/PyBaMM/pybamm/processed_variable.py in __init__(self, base_variable, t_sol, y_sol, mesh, interp_kind)
     52         # Evaluate the base_variable index-by-index
     53         for idx in range(len(t_sol)):
---> 54             entries[:, idx] = base_variable.evaluate(t_sol[idx], y_sol[:, idx])
     55 
     56         # assign attributes for reference

ValueError: could not broadcast input array from shape (9) into shape (10)
```

tinosulzer, 2019-05-02:
Will have a look, think I can fix this

tinosulzer, 2019-05-02:
There is a fundamental issue with evaluating concatenations of fluxes (we don't know how big each part should be, because it depends on the boundary conditions). I think we should change how the boundary conditions are done to fix this, see #335

PR #345 (2019-05-07): Issue 333 process particles
# Description

Change `ProcessedVariable` to allow considering 3D quantities, such as the concentration in the particles. Does *not* fix the fluxes problem, which is a bit deeper
Fixes #333 

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] Tests added that prove fi [... cut]
closed 2019-05-07 by tinosulzer