pybamm-team/PyBaMM #334: Add averaging to processed variable

Scottmar93, 2019-05-02:
**Summary**
Add a function to average quantities within `ProcessedVariable`. Mesh is already stored in `ProcessedVariable` so this should be a simple addition and it gets rid of a whole bunch of "Average" outputs.  

 **Motivation**
Don't need separator average variables which are useful for testing.

tinosulzer, 2019-05-02:
Issue with this is that the `Integrate` operator needs to be discretised

tinosulzer, 2020-06-15:
I think the workflow for this should be that the variable to be averaged should be added to the list of variables before the model is built and solved. This can be done in a script, for example:
```python3
model = pybamm.lithium_ion.DFN()
variable = model.variables["example variable"]
model.variables["X-averaged example variable"] = pybamm.x_average(variable)
...
```
Closing but feel free to reopen if different functionality is needed

closed 2020-06-15 by tinosulzer