pybamm-team/PyBaMM #362: Can we create KNOWN_DOMAINS dict instead of hard coding it?

Scottmar93, 2019-05-09:
Can we create the KNOWN_DOMAINS dictionary from the domains we enter in the geometry? 

For the add simple model notebook, I want to include a very simple model of SEI growth which I think covers a good range of useful things PyBaMM can do. But the domains currently KNOWN_DOMAINS aren't really applicable. It would be useful for the domains to be added when they are defined in the geometry.  

Maybe there is a half way of being able to add an additional domains whilst still allowing us to take advantage of the structure of domains already in KNOWN DOMAINS.

#### Motivation
I really like what has been done in having a `BaseModel` and a `BatteryBaseModel` as it allows other simple models to be created which don't need to follow a full battery structure, which is useful for adding simple models. I think we should have something similar for domains.

martinjrobins, 2019-05-09:
Yea, it would be nicer to have something more flexible than the current KNOWN_DOMAINS. At the moment all we use it for is for (I think) is DomainConcatenation, which takes a mesh. So you could have the list of KNOWN_DOMAINS contained in each geometry, and which is then contained within each mesh?

tinosulzer, 2019-05-10:
It's also used in `Symbol.domain` setter, but I'm not fussed about getting rid of that?

martinjrobins, 2019-05-10:
Yea, I don't think we need to check that in Symbol. Its probably something
the mesh or discretisation should check anyway

PR #372 (2019-05-12): Issue 362 known domains
# Description

Remove `pybamm.KNOWN_DOMAINS` and use mesh keys instead.
Fixes #362 

## Type of change

- [x] Breaking change (fix or feature that would cause existing functionality to not work as expected)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tests.py --quick`.

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] Tests added that pr [... cut]
closed 2019-05-12 by tinosulzer