pybamm-team/PyBaMM #375: Particle surface concentrations extrapolated below zero / above one

tinosulzer, 2019-05-10:
With the Finite Volumes, we get a `RuntimeWarning` in `Power` when we try to take the square root of a negative number. This is because the particle surface concentration in the exchange-current density is extrapolated from the bulk concentration, so can go below zero or above one even if the bulk concentration is within the limits.
(I think @jamiemfoster  has encountered this problem before). 

Options are:
1. Leave as is and accept the warning
2. Use more points for extrapolation (not obvious that this would work)
3. Cut off if extrapolated c hits the limits (from what I have seen, this only happens near the cut-off time anyway. But this might not be the case at higher C-rates.)
4. Accept as an issue for Finite Volumes and rely on other spatial methods to fix this
5. ... others?

My preference is for a combination of 3 & 4.

Scottmar93, 2019-05-10:
> Cut off if extrapolated c hits the limits (from what I have seen, this only happens near the cut-off time anyway. But this might not be the case at higher C-rates.)

I'm for this. Should be quite simple since we already have the surface values in the output dictionary.

jamiemfoster, 2019-05-10:
This is one, but not the only, reason to switch to finite elements/control
volumes. One reason to not be tempted by the cheap and dirty solution of
"cut it off" is that this introduces discontinuities in the system of ODEs
and you may find that it causes further issues down the line when time
stepping.

Scottmar93, 2019-05-10:
> One reason to not be tempted by the cheap and dirty solution of "cut it off" is that this introduces discontinuities in the system of ODEs

Ah, interesting. Does this also apply to using other "Events" like having a voltage cutoff etc? Or if we cut off when the internal concentrations reach zero and should we avoid cutting things off in general?

tinosulzer, 2019-05-10:
Not sure if we mean the same thing by "cut off", what I mean is terminate the time-stepping if this occurs (i.e. cut off in time rather than space). 
However, I guess it *might* be possible for this extrapolation problem to occur even if the bulk concentration is not close to zero/one (which is when we cut off already)?

jamiemfoster, 2019-05-10:
Not in the very same way, but they might share some similarities. The thing
which I believe is particularly obnoxious about the "cut it off" (by which
I mean "round it up to zero" or something similar) is that you might find
that when trying to take a time step close to an instance where we "round
it up to zero" the root finding step gets confused because because the
Jacobian appears singular; slightly perturbing the state vector \bf{u} does
not affect the RHS of the ODEs, \bf{f(u)}, and so you gets zeros in the
Jacobian where there shouldn't be. I am sure that this is soler dependent,
but I have found it to be a stumbling block with ode15s and so the issue
might be somewhat generic.

By contrast, things like approaching a cut-off voltage tend to be
reasonably smooth and so you might have to take smaller time steps to stop
exactly where you want to, but that's not normally too much of an issue.

-------------------------------------------------
Jamie M. Foster,
School of Mathematics and Physics,
University of Portsmouth,
Portsmouth, Hampshire, UK (PO1 2UP).
Phone: +44 02392 846 356 (internal: 6356)
Email: jamie.michael.foster@gmail.com
URL: http://ms.mcmaster.ca/~jmfoster/
-------------------------------------------------

PR #399 (2019-05-15): Issue 375 add event for surface concentration
# Description
Added an event for the surface concentration. Previous errors are resolved by putting in reasonable initial conditions for a charge. 

Fixes #375

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --uni [... cut]
closed 2019-05-15 by tinosulzer