pybamm-team/PyBaMM #478: Beta Testing

TomTranter, 2019-06-21:
Firstly, thank you for developing and making this software available. I am part of the Faraday Institute and was invited to become a Beta Tester. Please advise if this testing should be done on a particular development branch. I found the following bug on the master branch. 

**Describe the bug**
create_model example using scipy_solver throws error: UnboundLocalError: local variable 'termination' referenced before assignment.

If statements set the termination variable but there is no scenario to set it when sol.message is 'The solver successfully reached the interval end.'

**To Reproduce**
Steps to reproduce the behaviour:
1. Install master branch with no extra solvers
2. run create_model example

Kind regards,
Tom

martinjrobins, 2019-06-21:
Thanks @TomTranter, for reporting this. All the testing should be done on the master branch, as you have done. Individual issues we work on in separate branches, which are then merged into master once complete.

I'll have a look at this bug and get back to you.

TomTranter, 2019-06-21:
@martinjrobins in my experience it is better to a have a separate branch for development which is periodically merged into master which is kept for stable releases, all other branches can then be made from dev. Just a thought. I'd be happy to also fork the repository and make changes as I find bugs then make a pull request

martinjrobins, 2019-06-21:
Great, thanks @TomTranter, please do make a fork and PR when you are ready.

martinjrobins, 2019-06-23:
> @martinjrobins in my experience it is better to a have a separate branch for development which is periodically merged into master which is kept for stable releases,

Yes, I think this is a good suggestion that is worth thinking about in the future. At the moment we are in the situation where all the users are developers, so I think it is advantageous to keep everyone based off the one branch (i.e. master is dev). This might change as we get more users who want a stable release. Mind you, for stable releases we would probably want these to be obtained from PyPI rather than the git repo

TomTranter, 2019-07-02:
Hi @martinjrobins funnily enough I got this error on my laptop but installed it at work on a different machine and didn't get the error. Probably something in the python configuration. Incidentally I'm using windows and making an anaconda environment to run the code in. I've been exploring the code over the last few days and find it good but working with the variables is a little confusing. Also i'm interested in adding a new model or equation such as a joule heat source but it's not clear how to do this in the best way to get it solved automatically

martinjrobins, 2019-07-02:
by any chance is the version of python on your laptop 3.5 or lower? we are getting some numerical errors that pop up for python 3.5 (#382)

Thanks for the feedback, what do you mean that the variables are confusing? Do you mean the Variable class? Or the python objects you create while solving a model (i.e. the model, parameter_set, discretisation, solver etc)?

We have tried to document the process of creating a new model here: https://pybamm.readthedocs.io/en/latest/tutorials/add-model.html. If you want to add an extra equation to an existing model, it might be sufficient to simply create an existing model and update the relevant data structures (i.e. `self.rhs`, `self.algebraic`, `self.boundary_conditions`, `self.initial_conditions`, `self.variables`). If you need to discuss it with us, we have a pybamm group meeting on most Monday's which you are more than welcome to skype in on.

TomTranter, 2019-07-02:
My laptop could well be running 3.5 but I thought it was later. Will check.
By confusing I meant that actually inspecting and working with the outputs of the model was not intuitive to me. e.g. defining a ProcessedVariable and giving it a solution then calling it with a time and position etc. It feels like it should be a method of solution rather than a separate object but I'm sure there are very good reasons for this that I don't understand yet.
Thanks for your help and guidance so far. 
I think creating a whole new model for the heat source might be overkill. If I just want to multiply current by voltage and have that output and call it an Ohmic heat source then I'm not sure what the simplest way to do that is. I can do it completely separately using processedVariables but I thought there might be a way to add it as an extra variable in the model. Digging into the code a bit further maybe in the electrode.Ohm.get_explicit_leading_order method and similar methods it could be put there...

TomTranter, 2019-07-03:
Hi @martinjrobins I would like to join your next meeting as I think we have an opportunity to collaborate. My email is t.tranter@ucl.ac.uk if you want to send me details or maybe we can use a Faraday teams space

martinjrobins, 2019-07-03:
Fantastic, I'm not sure how to use the Faraday teams space, but happy to skype anyway. I'll send you my details. 

Btw. regarding adding a new output variable for a model, have a look at the "Output Variables" heading here: https://pybamm.readthedocs.io/en/latest/tutorials/add-model.html. You can put an arbitrary expression for the output variable that you define. This might be what you need?

closed 2019-07-19 by TomTranter